MotorolaMobilityLLC/kernel-msm #120: Moto E4 (perry) kernel source

aaronmarch, 2017-08-21:
Will the Moto E4 perry kernel source be released any time soon?

aaronmarch, 2017-09-22:
I know you guys are busy trying to release kernels for numerous devices and I really don't want to pester you, but it's been a while and I haven't even got a we're working on it or anything. Any word on the release of the Perry 7.1.1  NCQ.26.69-48/51 kernel source would be nice.

chenyt9, 2017-09-25:
Hi，aaronmarch，
    We are trying to release the code, perry the code has been in the workflow priority, please give us some time, we will be released before October 1. Thanks for understanding.
BRs.
chenyt

chenyt9, 2017-09-26:
Hi，@aaronmarch,
    Version MMI-NCQ26.69-48 is released:
    https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-NCQ26.69-48

closed 2017-09-26 by chenyt9