MotorolaMobilityLLC/kernel-msm #123: GPL Source Request for Moto G5s Plus (sanders) - (NPS26.116-26)

therootlord, 2017-09-05:
Hi, could you provide the corresponding sources ?

@charleseb @chenyt9

Many Thanks!

chenyt9, 2017-09-06:
Hi , therootlord ，

We have a process to follow before to release a version to Github. We are working to release Moto Z code, it will be available in a few days.

Thanks,

therootlord, 2017-09-06:
Great! 

Many thanks @chenyt9

CheckYourScreen, 2017-09-07:
@charleseb @chenyt9 any plans of releasing fastboot flashable imgs of stock retail firmware being shipped with the device?

therootlord, 2017-09-19:
Hi @chenyt9 ,  i really need this, to fix a f2fs mount issue, the sources from potter worked, but not camera/flash light/rear proximity sensor as its missing ltr559 driver,  ive imported this driver from another kernel source but not worked, can you provide a patch including the correct/missing camera/sensors drivers for sanders?

charleseb, 2017-09-19:
Hi @therootlord,

I'll working to release this version.

Charles

charleseb, 2017-09-20:
Version MMI-NPS26.116-45 released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-NPS26.116-45

closed 2017-09-20 by charleseb

CheckYourScreen, 2017-09-20:
Thanks @charleseb !

therootlord, 2017-09-21:
Many many thanks @charleseb  !

CheckYourScreen, 2017-11-25:
Hi @charleseb , can you please push the NPSS26.116-26-4 tag of sanders?