MotorolaMobilityLLC/kernel-msm #124: Moto Z2 Force Kernel Source NCX26.122-59

XirXes, 2017-09-06:
Can you please share kernel sources for Motorola Z2 Force (NASH) build NCX26.122-59?

http://www.gnu.org/licenses/gpl-violation.html

chenyt9, 2017-09-06:
Hi , XirXes,

We have a process to follow before to release a version to Github. We are working to release Moto Z code, it will be available in a few days.

Thanks,

XirXes, 2017-09-06:
Thanks for the clarification chenyt9. Glad to hear it.

thatcharacter, 2017-09-16:
Can we please get this source code now? It's been more than a few days.

techtony78, 2017-09-20:
+1, can we please have the kernel source code?

Thought it was supposed to be released last week. Moto Z2 Force

techtony78, 2017-09-22:
Hello?
This thing on? LOL

I thought HTC was bad, this is ridiculous.

charleseb, 2017-09-22:
Hi all,

We are working to release this version soon.

Thanks,
Charles

techtony78, 2017-09-23:
Right. Empty promises as this same thing was said a couple weeks back to shut us up.

slipkn0t2012, 2017-09-24:
How soon, is soon Charles? What's an ETA?

charleseb, 2017-09-25:
Hi @slipkn0t2012,

I'll release this version until next Friday.

Thanks,
Charles

Uzephi, 2017-09-26:
So if I am reading the broken English correctly, (not blaming you, your native language might not be English and it is the translator error) you will have the kernel released NLT (No later than) 06/10/2017 dd/mm/yyyy? 

I.E. Next Friday the code will be released?

charleseb, 2017-09-27:
Hi @Uzephi,

"by Friday"

Ch.

charleseb, 2017-09-27:
Version MMI-NCX26.122-59-8 is released.
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-NCX26.122-59-8

closed 2017-09-27 by charleseb

XirXes, 2017-09-27:
Awesome! Feeling more confident in my choice to switch to Moto now.