MotorolaMobilityLLC/kernel-msm #126: Moto Z (griffin)

samstone86, 2017-09-25:
@charleseb Hi could you please release an  devicespecific tree for griffin?

can you may also tell me if softap is mandatory for getting propper ril build for griffin? (contact mail is visible in my profile if you want to answer this way)

awheeler-mmi, 2017-09-25:
Hey @samstone86, we don't release the non-GNU source like the device configuration, and aren't in a good position to support system builds.  But we definitely appreciate the interest!  I recommend strolling over to XDA for this kind of discussion.  FWIW, I am not aware of any RIL dependencies on softAP.

Good luck!

-andrew

closed 2017-09-25 by awheeler-mmi

luk1337, 2017-09-25:
@awheeler-mmi as far as I can tell older Moto devices depend on wifi_qsap_set_tx_power function in softap (system/qcom). I don't know if that's still a thing with newer devices such as Moto Z.

erfanoabdi, 2017-09-25:
Yep, here on griffin we need wifi_qsap_set_tx_power on softap to get ril work.
I think Potter also need this.

_Sent from my Motorola XT1650-05 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

awheeler-mmi, 2017-09-25:
Ok, yup, I found the reference - thanks for the pointer to the function.  One of our proprietary shared RIL-ish libs depends on softap.