MotorolaMobilityLLC/kernel-msm #130: Kernel Source Request for XT1676 (Moto G5 / Cedric) NPP25.137-15/13

Aerolivier, 2017-11-02:
Hello,

Please could you provide the kernel sources for the Moto G5 (3GB RAM variant, **XT1676**) (codenamed Cedric), with tag **NPP25.137-15**.

I understand that some sources for cedric have already been released, but they have a different version number and we are running into some issues with them on the XT1676 devices.

Many thanks, and I hope you are well.

charleseb, 2017-11-09:
Hi @Aerolivier,

We are working to release this version. It will be released by Tuesday.

Thanks,
Charles

m32, 2017-11-23:
Could you provide updates for xt1676 npps25.137-72-4

charleseb, 2017-11-23:
Version released  
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-NPP25.137-15

Aerolivier, 2017-11-23:
@charleseb Many thanks!

closed 2017-11-23 by Aerolivier