MotorolaMobilityLLC/kernel-msm #131: e4 Plus kernel

jaduncan, 2017-11-20:
Do you have a timescale for release of the e4 plus/XT17xx/nicklaus kernel source?

For the avoidance of doubt I'm using the device, and this is a formal GPL request for a source link.

samstone86, 2017-11-20:
https://github.com/MotorolaMobilityLLC/kernel-mtk/tree/nougat-7.1.1-release-nicklaus ?

archon810, 2017-11-21:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-NCR26.58-26
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-NPR26.58-25
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-NDR26.58-21

hangfan1, 2017-12-01:
Moto E Plus
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-NCR26.58-26
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-NPR26.58-25
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-NDR26.58-21

closed 2017-12-01 by hangfan1