MotorolaMobilityLLC/kernel-msm #134: USB Soundcard sometimes not recognized

Loader009, 2017-11-29:
I've got an issue with my USB soundcard, bought here:
https://hifimediy.com/Sabre-9018-Type-C-DAC 
mine is an older batch but after contacting the seller, they said that my problem wouldn't be solved by the new batch.

Firmware version: NPLS26.118-20-5-3
Kernel: (from erfanoabdi) current TurboZ_v1.6_7.1.1_StockN
From here: https://forum.xda-developers.com/moto-z/development/kernel-turbo-z-custom-kernel-moto-z-t3493354
(problem also existent with stock kernel)

Problem:
When I connect the soundcard, it is sometimes recognized (red LED is and stays on) and sometimes not (right after plugging if the red LED flashes for a moment and stays then dark). It might work after replugging it a dozen times but I haven't found any hint on to get recognition reliable.

Also (and that's kind of weird) sometimes if I wiggle the type-c plug a bit (e.g. while moving the phone), the soundcard (while recognized) gets reconnected. But only sometimes. Sometimes I don't need to worry about it, the connection is stable all the time.

I tested on my PC and an Asus Nexus 7 (2013), no such problems so far.
I also tested on two different Samsung devices (Galaxy S6 or newer), it worked instantly.
Because of this I don't think my soundcard is defective. (No "reconnect" issue here.)

I did some dmesg logs, this is what you need to look at:
successfully recognized
`[  521.178183,1] usb 1-1: new full-speed USB device number 2 using xhci-hcd
[  521.342506,1] usb 1-1: config 1 has an invalid interface number: 3 but max is 2
[  521.342554,1] usb 1-1: config 1 has an invalid interface number: 3 but max is 2
[  521.342571,1] usb 1-1: config 1 has an invalid interface number: 3 but max is 2
[  521.342586,1] usb 1-1: config 1 has an invalid interface number: 3 but max is 2
[  521.342601,1] usb 1-1: config 1 has no interface number 2
[  521.359624,1] usb 1-1: New USB device found, idVendor=262a, idProduct=10e3
[  521.359670,1] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  521.359686,1] usb 1-1: Product: SA9023 USB Audio
[  521.359700,1] usb 1-1: Manufacturer: HiFimeDIY Audio
[  521.410561,1] input: HiFimeDIY Audio SA9023 USB Audio as /devices/soc/6a00000.ssusb/6a00000.dwc3/xhci-hcd.0.auto/usb1/1-1/1-1:1.0/0003:262A:10E3.0001/input/input14
[  521.412437,1] hid-generic 0003:262A:10E3.0001: input,hidraw0: USB HID v1.00 Device [HiFimeDIY Audio SA9023 USB Audio] on usb-xhci-hcd.0.auto-1/input0`

and failed
`[  537.012756,0] hub 1-0:1.0: USB hub found
[  537.013008,0] hub 1-0:1.0: 1 port detected
[  537.013788,0] xhci-hcd xhci-hcd.0.auto: xHCI Host Controller
[  537.015498,0] xhci-hcd xhci-hcd.0.auto: new USB bus registered, assigned bus number 2
[  537.016046,0] usb usb2: New USB device found, idVendor=1d6b, idProduct=0003
[  537.016072,0] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1`

Please tell me if you need anything else and what I can test to help to solve this.

dmesg logs:
[1. before connected.txt](https://github.com/MotorolaMobilityLLC/kernel-msm/files/1514911/1.before.connected.txt)
[2. success.txt](https://github.com/MotorolaMobilityLLC/kernel-msm/files/1514913/2.success.txt)
[3. fail.txt](https://github.com/MotorolaMobilityLLC/kernel-msm/files/1514914/3.fail.txt)

awheeler-mmi, 2017-12-01:
Hey @Loader009, I opened a bug w/ our development staff.  If we apply a fix then it would come out with a (general) upgrade, not a security update.  So keep your eye out for it when your device gets upgrades.

Thanks!

-andrew

closed 2017-12-01 by awheeler-mmi

awheeler-mmi, 2017-12-07:
@Loader009 keep your eye out for [Change-Id I0bca9e78a0158d71423e6a767e63a1ffa35882bf](https://github.com/MotorolaMobilityLLC/kernel-msm/search?utf8=%E2%9C%93&q=I0bca9e78a0158d71423e6a767e63a1ffa35882bf&type=) -- the kernel team is telling me that it should fix it.  It may not go to Vector first.

Loader009, 2018-03-28:
Have this fix been merged to the oreo-8.0.0-release-griffin branch?
I have waited for a new release but sadly I still have this problem with the oreo kernel.

I haven't seen the Change-Id here yet, so I guess it never got published.