MotorolaMobilityLLC/kernel-msm #139: f2fs patches

GetOffMyLawnNow, 2018-01-06:
Would it be possible to add the following f2fs kernel patches in the next Owens update so we don't have to keep rebuilding the kernels? It has been added to some other vendor repositories.
https://git.kernel.org/cgit/linux/kernel/git/jaegeuk/f2fs-stable.git/commit/?h=linux-3.18.y&id=0d5ed69805ee57d7b416c4a6b47c3b67284db105
https://git.kernel.org/cgit/linux/kernel/git/jaegeuk/f2fs-stable.git/commit/?h=linux-3.18.y&id=7a895023cc2c06c08a6cccb71e112255eb0837ee

Deadlock patch:
https://git.kernel.org/cgit/linux/kernel/git/jaegeuk/f2fs-stable.git/commit/?h=linux-3.18.y&id=7a895023cc2c06c08a6cccb71e112255eb0837ee
Thanks.

GetOffMyLawnNow, 2018-01-06:
Actually, adding these for all devices would be great.

awheeler-mmi, 2018-01-08:
Hey @GetOffMyLawnNow, I opened a bug for our kernel team, and pointed them to this issue.  Thanks!!

closed 2018-01-08 by awheeler-mmi

GetOffMyLawnNow, 2018-01-08:
No. Thank you!

GetOffMyLawnNow, 2018-01-08:
*Np

GetOffMyLawnNow, 2018-01-10:
I meant to add that the 3rd commit listed needs to be reverted, if f2fs is upstreamed to 4.10.  Also, I am not the one who found these patches. It was arter97 who pointed them out here: https://forum.xda-developers.com/showpost.php?p=70492947&postcount=8793

luca020400, 2018-01-11:
I'm not really a fun of these /reverts/, now that Jaegeuk Kim is working for/whit google on f2fs for Android I would just rather stick with the kernel_common fs/f2fs ( even though we recently had to revert the latest changes because they were causing kernel panics, but ¯\_(ツ)_/¯ )