MotorolaMobilityLLC/kernel-msm #143: Moto G5 Plus "Potter" - NPN25.137-83 Kernel Sources

attack11, 2018-02-06:


charleseb, 2018-02-07:
Hi @attack11,

We are working to release this version by Wednesday.

-Charles

9jkh, 2018-02-09:
Got it OTA few days ago for G5+ 

Does this update patch KRACK, Bluborne, Spectre-Meltdown?

Ristovski, 2018-02-15:
@charleseb Any progress on this?

charleseb, 2018-02-15:
Hi @Ristovski,

I'm working to conclude this push.

Thanks,
Charles

charleseb, 2018-02-15:
Version released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-NPN25.137-93

Thanks,
Charles

closed 2018-02-15 by charleseb

awheeler-mmi, 2018-02-15:
He @9jkh,

Good to hear you got the update.  The Bluborne patches came in with the September patch set, so you are good there.  The KRAK and Spectre/Meltdown fixes are part of the January patch set, which are in the queue for the next update.  But potter is A53, so it's not particularly susceptible to Spectre or Meltdown.

Jdushey, 2018-03-21:
I just purchased a Moto G5 Plus - XT1687.
1. Looking for stock image to flash in case something happens while trying to root the device. 
2. Is Oreo available for this device ?

awheeler-mmi, 2018-03-21:
Hey @Jdushey, 

We put all released factory images [here](https://motorola-global-portal.custhelp.com/app/standalone/bootloader/recovery-images).  We have not released images for the G5 Plus, but they appear to be available if you look around.

The G5 Plus Oreo release is coming!  Stay tuned [here](https://mobilesupport.lenovo.com/us/en/softwareupgrade) for more info.

Thanks,

-andrew