MotorolaMobilityLLC/kernel-msm #146: Source code release

Goose03, 2018-03-05:
Is there any chance we can get the latest kenel source code released for Owens and Perry? Please, and thank you!

luk1337, 2018-03-05:
You need to provide the exact versions otherwise you'll get ignored.

charleseb, 2018-03-05:
Hi @Goose03,

Could you describe the version you need, please?

Thanks,
Charles

Goose03, 2018-03-06:
MMI-NPR26.58-45 for Owens
MMI-NCQ26.69-56 for Perry
Thank you very much!

Goose03, 2018-03-08:
Not a problem, I know you guys have a lot to deal with. That was expected. Thank you very much for your quick response.

chenyt9, 2018-03-12:
@Goose03 , Versions have been released.
MMI-NPR26.58-45 for Owens:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-NPR26.58-45
MMI-NCQ26.69-56 for Perry:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-NCQ26.69-56
Thanks.

closed 2018-03-12 by chenyt9