MotorolaMobilityLLC/kernel-msm #147: Moto Z "Griffin" - OCC27.102-18 Kernel Sources

erfanoabdi, 2018-03-20:
Hi @charleseb @chenyt9 @awheeler-mmi

Blur Version: Blur_Version.27.11.22.griffin_retcn.retcn.en.cn

Could you provide the corresponding sources(v4l2, greybus, kernel) ?

Thanks in advance!

hangfan1, 2018-03-23:
Hi erfanoabdi
I will push the sources code for version OCC27.102-18 to github in one weeks.
thanks
Hangfan

hangfan1, 2018-03-27:
Hi erfanoabdi
I has pushed the sources code for version OCC27.102-18.
The tag is MMI-OCC27.102-18 and readme file is MMI-OCC27.102-18.txt.

thanks
Hangfan

closed 2018-03-27 by hangfan1

erfanoabdi, 2018-03-27:
Thanks @hangfan1 for kernel source, but i can't see tag MMI-OCC27.102-18 for greybus, v4l2_hal and **qcacld2**
Could you please also provide sources for following repos : 
https://github.com/MotorolaMobilityLLC/motorola-external-greybus
https://github.com/MotorolaMobilityLLC/motorola-external-v4l2_hal
https://github.com/MotorolaMobilityLLC/vendor-qcom-opensource-wlan-qcacld-2.0

hangfan1, 2018-03-27:
Hi erfanoabdi
Sorry for missed vendor-qcom-opensource-wlan-qcacld-2.0. I will push this tag.
However, Checked in the full build log, the project(Moto Z "Griffin") don't use motorola-external-greybus and motorola-external-v4l2_hal .

thanks
Hangfan

erfanoabdi, 2018-03-27:
could you please check build log again, griffin have moto mods feature and kernel should have greybus modules for connecting to mods in kernel side and v4l2 for camera mod
also in stock rom we can see : 
```
vendor/lib/modules/gb-audio.ko
vendor/lib/modules/gb-battery.ko
vendor/lib/modules/gb-camera_ext.ko
vendor/lib/modules/gb-display.ko
vendor/lib/modules/gb-hid.ko
vendor/lib/modules/gb-light.ko
vendor/lib/modules/gb-mods.ko
vendor/lib/modules/gb-ptp.ko
vendor/lib/modules/gb-raw.ko
vendor/lib/modules/gb-sensors_ext.ko
vendor/lib/modules/gb-usb_ext.ko
vendor/lib/modules/gb-vendor-moto.ko
vendor/lib/modules/gbsim-mods-sim.ko
vendor/lib/modules/greybus.ko
vendor/lib/modules/v4l2-hal.ko
```

Thanks

hangfan1, 2018-03-27:
Hi 
I will take a full build again .
If the project used these two module, I wil push these to github.
thanks for reminding.

Hangfan

hangfan1, 2018-03-28:
Hi erfanoabdi
Thanks for reminding.
I have pushed the tag to 
https://github.com/MotorolaMobilityLLC/motorola-external-greybus
https://github.com/MotorolaMobilityLLC/motorola-external-v4l2_hal

thanks
Hangfan

erfanoabdi, 2018-03-28:
thank you very much ❤️, i'm still waiting for qcacld driver

hangfan1, 2018-03-28:
Hi
Tag MMI-OCC27.102-18 has push to
https://github.com/MotorolaMobilityLLC/vendor-qcom-opensource-wlan-qcacld-2.0 
thanks
Hangfan