MotorolaMobilityLLC/kernel-msm #148: Request for NPSS26.116-61-5 tag source update of sanders

Keertesh19, 2018-03-29:
Please release source update for NPSS26.116-61-5 tag of Moto G5S Plus (sanders).

Keertesh19, 2018-03-29:
@chenyt9 @hangfan1

Keertesh19, 2018-03-29:
And please release/mail balnk flash files for snaders if possible. Thanks! 😁

CheckYourScreen, 2018-03-29:
@awheeler-mmi many users reported loss of imei/volte after flashing of latest firmware(NPS26.116-61/NPSS26.116-61-2) having new bootloader(C2.07) and updated fsg/modem/firmware.
I cant figure out what's the exact cause of this issue, will be great if you could shed some light on the fix or maybe release blankflash files for sanders to try out a clean flash.

CheckYourScreen, 2018-04-04:
@chenyt9 @charleseb @awheeler-mmi @hangfan1 any update on this?

CheckYourScreen, 2018-04-12:
Hi @charleseb ,
See if you can push april patch(upcoming) as well along with the current tag you are preparing to release.
New OTA should be out in few days so it will save time instead of waiting again for a month.
Next one for sanders should be of oreo right?

charleseb, 2018-04-12:
Hi @CheckYourScreen,

This version will be released by Monday.

Thanks,
Charles

daudibrahim131, 2018-04-13:
Sir please release ROM of NPSS26.116-61-5 my Moto g5s plus have lost it's volte feature and ask for only stock after this ROM .

thirunko, 2018-04-16:
please release blankflash file for moto g5s xt1792

CheckYourScreen, 2018-04-18:
Hi @charleseb ,
Which monday? :)

CheckYourScreen, 2018-04-23:
Hi @charleseb @chenyt9 ,
Any update on this?

charleseb, 2018-04-23:
Hi all,

Sorry. I was out for a while. I'm working to push the version this week.

Thanks,
Charles

charleseb, 2018-04-26:
Version released:

https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-NPSS26.116-61-5

closed 2018-04-26 by charleseb

CheckYourScreen, 2018-04-26:
Thanks @charleseb !
Can you plz check for NPSS26.116-61-8 for being in pipeline? I heard some users are getting april patch with that build number, anyways thanks again!

charleseb, 2018-04-26:
Hi @CheckYourScreen,

I'm starting the process to release this version.

Thanks,
Charles

thirunko, 2018-05-01:
Hi @charleseb, 
you can make the blankflash file available to recover this device Moto G5S xt1792

CheckYourScreen, 2018-05-04:
Hi @charleseb ,
Any eta for NPSS26.116-61-8 tag?
Thanks!