MotorolaMobilityLLC/kernel-msm #152: Kernel source for March update Nash OCX27.109-48-3 and OCX27.1.2

Uzephi, 2018-05-06:
Since the March update should have the January, February and March security updates and the CAF kernel updates, can we get that source? Upon looking at the kernel files side by side in a binary checker, there isn't much of a change between the two and I would like to see if those security patches are in there. The one patch for KPTI leaks in 4.4.112 (used in CAF as well) actually breaks motomod functionality and I would love to see your diff so I can put it in my kernel as well. https://github.com/Uzephi/kernel_nash/commit/b078041b87c6eebd41cac385cc6216c0f6144931 for reference.

I mean I would love to believe I can trust my oem releasing a security update that has the actual patches, but the binary files are currently saying otherwise.

Uzephi, 2018-05-06:
Below is the example explained above where the code just shifted addresses, possibly due to the empty block of the uncompressed image.

[image]
[image]

chenyt9, 2018-05-29:
The latest payton tag is: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-OPWS27.57-43-2-4-3
That is, the source code of OPWS27.57-43-2-3 version.

erfanoabdi, 2018-05-29:
thanks for new payton tag but any chances on OPW28.1 payton kernel source ?

Uzephi, 2018-06-17:
Any update on officially having the Nash tags? We also got another update recently. 27.109-51-4 for T-Mobile. 27.109-48-6 for Sprint. Any ETA on when source for these will be released?

I hate to link this but... https://www.gnu.org/licenses/gpl-violation.en.html

chenyt9, 2018-06-21:
@Uzephi,
I'll check it. But sorry, I need to confirm that the version you requested is OCX27.109-51-4 and OCX27.109-48-6, right?
Thanks.

Uzephi, 2018-06-21:
@chenyt9 yes. Thank you for the prompt response. I am glad to get an update on what is going on. I understand the process of how long it can take. I just got frustrated because -3 and 27.1.2 were promised about a month ago and still waiting for it. I have been holding off on my kernel release since it is still running the December Kernel and doesn't have anything from you on March or the newer update.

chenyt9, 2018-06-25:
@erfanoabdi ,
Could you specify the version you need in more detail? Thanks.

chenyt9, 2018-06-25:
@Uzephi ,
Because we have some necessary processes, so we need a little time. We will release it ASAP. Usually, it will be released within two weeks.
Thanks for understanding
BRs.
chenyt9

erfanoabdi, 2018-06-25:
@chenyt9 thanks for reply
yeah, sure

Payton
Build Id: OPW28.46-13
Build Fingerprint: motorola/payton/payton:8.1.0/OPW28.46-13/AQZXUTus:user/release-keys
Blur Version: Blur_Version.28.11.16.payton.retail.en.US

Uzephi, 2018-07-03:
https://androidfilehost.com/?fid=674106145207492373
NASH_SPRINT_OCXS27.109-48-6_subsidy-DEFAULT_regulatory-DEFAULT_CFC
https://androidfilehost.com/?fid=818222786056035860
NASH_TMO_C_OCXS27.109-51-4_subsidy-TMO_RSU_regulatory-DEFAULT_CFC

Link to firmware and name of firmware files. Sorry, I haven't ran stock in a long time to grab those in a timely manner.

chenyt9, 2018-07-03:
@Uzephi,
I have to re-apply for permission to publish. The previously applied version is not OCXS.

chenyt9, 2018-07-04:
@Uzephi,
Version Released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-OCXS27.109-51-4
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-OCXS27.109-48-6
Thanks.

Uzephi, 2018-07-04:
Thank you very much.

closed 2018-07-04 by Uzephi

chenyt9, 2018-07-06:
@erfanoabdi，
Version Released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-OPW28.46-13
Thanks.

erfanoabdi, 2018-07-06:
Awesome, thank you so much