MotorolaMobilityLLC/kernel-msm #153: Moto G6 Plus "Evert" - OPW27.113-45 Kernel Sources

ahmedalbayati, 2018-06-11:
Hi, @charleseb @chenyt9 @awheeler-mmi @hangfan1 

**Blur Version:** Blur_Version.27.11.53.evert.retail.en.US
**Build Fingerprint:** motorola/evert/evert:8.0.0/OPW27.113-45/53:user/release-keys
**Android Version:** 8.0.0

Can you provide the corresponding sources?
Thanks in advance!

chenyt9, 2018-06-12:
@ahmedalbayati， @creasingmass ，
Because we have some necessary processes, so we need a little time. We will release OPW27.113-45 ASAP. It will be released within two weeks.
Thanks for understanding
BRs.
chenyt9

charleseb, 2018-06-12:
Hi guys,

We are working to release this version. It will be available by Friday.

Thanks,
Charles

chenyt9, 2018-06-14:
@creasingmass ，@ahmedalbayati,
Version released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-OPW27.113-45
thanks.

closed 2018-06-14 by chenyt9