MotorolaMobilityLLC/kernel-msm #154: Moto G6 "Ali" - OPS27.104-15-10 Kernel Sources

sir-maniac, 2018-06-13:
Hi, @charleseb @chenyt9 @awheeler-mmi @hangfan1

Blur Version: Blur_Version.27.51.12.ali.retail.en.US
Build Fingerprint: motorola/ali/ali:8.0.0/OPS27.104-15-10/12:user/release-keys
Android Version: 8.0.0

Can you provide the corresponding sources?
Thanks in advance!

chenyt9, 2018-06-21:
@sir-maniac ,
Because we have some necessary processes, so we need a little time. We will release OPS27.104-15-10 ASAP. It will be released within one weeks.
Thanks for understanding.
BRs.
chenyt9

sir-maniac, 2018-06-21:
@chenyt9  Thank you.

chenyt9, 2018-06-25:
@sir-maniac,
Version released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-OPS27.104-15-10
Thanks.

closed 2018-06-25 by chenyt9