MotorolaMobilityLLC/kernel-msm #155: Request for OPS28.49-2 tag source update of sanders

Keertesh19, 2018-06-16:
Please release source update for OPS28.49-2 tag source update of sanders

Keertesh19, 2018-06-16:
@charleseb @chenyt9 @hangfan1

charleseb, 2018-07-23:
Hi @Keertesh19,

We are working to release this version. It will be released by Friday.

Thanks,
Charles

CheckYourScreen, 2018-07-25:
@charleseb could you please release the latest tag (OPS28.65-14) instead.
Thanks!

CheckYourScreen, 2018-07-30:
@charleseb @chenyt9 any update on this?

chenyt9, 2018-07-31:
@Keertesh19 ，
Version Released：
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-OPS28.49-2
Thanks.

attack11, 2018-08-21:
@chenyt9 missing the f2fs "motorola-kernel" sources from the released Tag OPS28.49-2

chenyt9, 2018-08-22:
@attack11 ，
Thank you for your reminder, it has been added.
Please refer to: https://github.com/MotorolaMobilityLLC/motorola-kernel/releases/tag/MMI-OPS28.49-2

attack11, 2018-08-22:
Thanks @chenyt9 , One last thing,Can you release the readme please?

chenyt9, 2018-08-29:
@attack11 ,
https://github.com/MotorolaMobilityLLC/readme/blob/master/MMI-OPS28.49-2.txt
Thanks.

closed 2018-08-29 by chenyt9

chenyt9, 2018-10-12:
@CheckYourScreen 
Version released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-OPS28.65-14
Thanks.

CheckYourScreen, 2018-10-12:
Thanks @chenyt9