MotorolaMobilityLLC/kernel-msm #156: Oreo 8.0.0 James Source

codyf86, 2018-07-01:
Please release the source for the james kernel.  Was able to boot the LA.UM.6.6 codeaurora kernel, but battery stats and cell radio do not work.

chenyt9, 2018-07-03:
@codyf86 ，
Please provide a specific version number, I will apply for release, thank you!

codyf86, 2018-07-03:
OCP27.91-23 :)

codyf86, 2018-07-04:
Also, a shot in the dark, but if you could manifest this file from motorola JAMES_SPRINT_OCP27.91-23_cid50_XT1921-5-SS_8.0.0_subsidy-DEFAULT_regulatory-DEFAULT_CFC.xml.zip

I'm the only one developing for the e5 atm and managed to soft brick one already haha. :)  Had to go buy a new one.  If it's possible email me codyf86@gmail.com I know the files been released but its on like one site that wants you to pay a membership to download it with no easy way to do that. All good either way. :)

chenyt9, 2018-07-10:
@codyf86 
Version released.
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-OCP27.91-23
Thanks.

closed 2018-07-10 by chenyt9