MotorolaMobilityLLC/kernel-msm #157: Moto Z Play OPN27.76-12-22 / OPNS27.76-12-22-3 kernel sources

TimoRoth, 2018-07-12:
I'm looking for the Kernel of one of the following two addison oreo releases:

Build Fingerprint: motorola/addison/addison:8.0.0/OPNS27.76-12-22-3/3:user/release-keys
Blur Version: Blur_Version.27.201.3.addison.retail.en.US
Android Version: 8.0.0

or:

Build Fingerprint: motorola/addison/addison:8.0.0/OPN27.76-12-22/24:user/release-keys
Blur Version: Blur_Version.27.21.24.addison.retail.en.US
Android Version: 8.0.0

Pretty much the latest Kernel available for addison Oreo

BtbN, 2018-07-14:
@charleseb @chenyt9 @awheeler-mmi @hangfan1

charleseb, 2018-07-23:
Hi @TimoRoth, @BtbN,

We are working to release this version. It will be released by Friday.

Thanks,
Charles

chenyt9, 2018-08-03:
@TimoRoth @BtbN
Version released.
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-OPNS27.76-12-22-24-3
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-OPN27.76-12-22
Thanks.

closed 2018-08-03 by chenyt9