MotorolaMobilityLLC/kernel-msm #163: Please provide VoLTE Update patch for Sanders for Airtel and Vodafone

mg5sp, 2018-08-09:
Please provide update patch to enable VoLTE for Bharti Airtel and Vodafone in sanders.At present VoLTE is supported only for Jio.But other moto phones like E4 has rolled out airtel volte update, also recently launched G6 series supports VoLTE for other carriers.

mg5sp, 2018-08-09:
@charleseb @chenyt9 @awheeler-mmi @hangfan1

awheeler-mmi, 2018-08-09:
Hey @mg5sp,  We enable VoLTE per-device and per-carrier, based on agreements with the carriers, and based on our performing the required testing.  So we don't provide provide patches to enable VoLTE on new carriers, as that would essentially bypass our agreements and required testing.  We sometimes do enable VoLTE on carriers during an upgrade, so keep your eye out after taking an (non-security-only) update.

closed 2018-10-13 by awheeler-mmi