MotorolaMobilityLLC/kernel-msm #164: Moto E4 (perry) kernel source is broken

RebelLion420, 2018-08-17:
Source wont compile without a lot of rewrite, is there any way you could release a complete and functional source? It would be much appreciated for a device that seems to be no longer officially supported but valuable nonetheless

jasonmerc, 2018-09-19:
As @motorolamobility  may be aware, releasing incomplete/broken kernel source is a violation of the GPL, which is treated as a "contract" of sorts.  As such, failure to comply is breaking the law.

Hopefully they'll come through.

chenyt9, 2018-09-20:
@RebelLion420 @jasonmerc ,
I double confirmed that the kernel code of our Perry project is complete and can be compiled. Which version do you get and the compilation failed?
I suggest that you can get the corresponding version compile instructions from the following URL：
https://github.com/MotorolaMobilityLLC/readme
e.g.
For MMI-NPQ26.69-41:
https://github.com/MotorolaMobilityLLC/readme/blob/master/MMI-NPQ26.69-41.txt

Thanks.

RebelLion420, 2018-09-22:
I actually found the proper instructions thanks to @nathanchance and got it to compile correctly after adding in the f2fs files and generating the defconfig. I'm closing this issue now than, thanks

closed 2018-09-22 by RebelLion420