MotorolaMobilityLLC/kernel-msm #165: Beckham Moto z3 play USA motorola.com model

Motophan, 2018-08-18:
Please release kernel source for this. Thanks for your time. 

@charleseb @chenyt9 @awheeler-mmi @hangfan1

charleseb, 2018-08-20:
Hi @Motophan,

Please inform the version you need.

Thanks,
Charles

Motophan, 2018-08-22:
> OPW28.70-25

There is only one version to my knowledge. This is a presale question. The
BLU (zte) and OP3/5 (One plus) had spyware embedded in the phone, only to
be discovered with source code release. Moto is a honest company, but I
will not buy a phone unless it source is available. the Moto z3 play is on
my short list because of 1- microsd, 2- unlockable bootloader (I am coming
from a Moto x pure, a phone I loved) 3- game controller add-on $79.99.
Those make this phone very compelling. My Moto X Pure was a very good
phone, but only gets about an hour of battery life because I used it every
day since 2015. Best purchase I have ever made, learned how to program for
Android with it. The x4 is too small. The g6 does not have modular
capabilities. I will buy one for me and my wife once source code is
available. Thank you.

Motophan, 2018-08-22:
OPW28.70-25

charleseb, 2018-08-22:
Hi @Motophan,

I'm working to release this version. It will be available by Tuesday.

Thanks,
Charles