MotorolaMobilityLLC/kernel-msm #172: What's the bluetooth stack being shipped on MMI-MPJ24.139?

thuytrinh, 2018-09-29:
I'm just wondering whether the bluetooth stack being shipped (or used) on [MMI-MPJ24.139](https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-MPJ24.139) is [bluez](https://android.googlesource.com/platform/external/bluetooth/bluez/) or [bluedroid](https://android.googlesource.com/platform/external/bluetooth/bluedroid/)? I'm investigate a strange issue with bluetooth LE on a Moto G4 Plus running `MMI-MPJ24.139` and I couldn't see how that strange issue could happen with `bluedroid`. So probably the `MMI-MPJ24.139` is using `bluez` instead?

Thanks for any response. Sorry if my question sounds stupid.

awheeler-mmi, 2018-10-13:
Hey @thuytrinh, are you asking about athene?  I believe athene was using the CAF bluetooth stack, rather than bluez or bluedroid.  I can double-check with our Bluetooth team next week.

thuytrinh, 2018-11-17:
Sorry, I think my question was quite misleading.

closed 2018-11-17 by thuytrinh