MotorolaMobilityLLC/kernel-msm #176: Request for OPWS28.46-21-12   PAYTON_FI (moto x4)  kernel sources

kraatus90, 2018-10-11:
Can we have monthly kernel updates according to security patches please ?

kraatus90, 2018-10-12:
@charleseb

awheeler-mmi, 2018-10-13:
Hey @kraatus90, we're getting the releases out as soon as possible.   A number of open-source-release reviews just want across my (virtual) desk.  I don't recall if this payton release was one of them, but @charlesb will respond with an ETA next week.  

FYI, we prioritize the tags that developers are asking for, so your formal requests here on Github help!!!

charleseb, 2018-10-18:
Hi @kraatus90,

This version will be released by Tuesday.

Thanks,
Charles

charleseb, 2018-10-23:
Version released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-OPWS28.46-21-12

closed 2018-10-23 by charleseb