MotorolaMobilityLLC/kernel-msm #182: Error when compiling OPP28.85-16 - possible faulty makefile.

ezxpro, 2018-10-30:
I tried with several toolchains, always get the same error
`  CC      block/blk-lib.o
  EXTRACT_CERTS   ./""
At main.c:154:
- SSL error:0906D06C:PEM routines:PEM_read_bio:no start line: ../crypto/pem/pem_lib.c:691
extract-cert: ./: Is a directory
rm: cannot remove 'certs/x509_certificate_list': No such file or directory
certs/Makefile:26: recipe for target 'certs/x509_certificate_list' failed
make[1]: *** [certs/x509_certificate_list] Error 1
Makefile:967: recipe for target 'certs' failed
make: *** [certs] Error 2
make: *** Waiting for unfinished jobs....
  CC      block/blk-mq.o
  CC      block/blk-mq-tag.o
  CC      block/blk-mq-sysfs.o
  CC      block/blk-mq-cpu.o
  CC      block/blk-mq-cpumap.o
  CC      block/ioctl.o
  CC      block/genhd.o
  CC      block/scsi_ioctl.o
  CC      block/partition-generic.o
  CC      block/ioprio.o
  CC      block/partitions/check.o
  CC      block/partitions/msdos.o
  CC      block/partitions/efi.o
  CC      block/bounce.o
  CC      block/bsg.o
  LD      block/partitions/built-in.o
  CC      block/noop-iosched.o
  CC      block/deadline-iosched.o
  CC      block/cfq-iosched.o
  CC [M]  block/test-iosched.o
`

So I checked the the line 26 of certs/Makefile, as pointed by make, and it says: 
`$(call if_changed,extract_certs,$(srctree)/$(CONFIG_SYSTEM_TRUSTED_KEYS))`
but I don't know exactly what that means.
It also seems that there's a problem in Lines 17 and 18
`quiet_cmd_extract_certs  = EXTRACT_CERTS   $(patsubst "%",%,$(2))
      cmd_extract_certs  = scripts/extract-cert $(2) $@ || ( **rm** $@; exit 1)`
seems like the highlighted rm command tries to remove a directory that doesn't exist.

I don't know what exactly is happening but this receipt is failing everytime. I'm currently using a x86_64 Linux host, but I also tried to compile it using ARM's bare metal compiler on an ARM host and compilation fails in the same place.

ezxpro, 2018-10-30:
@jwylder1 @awheeler-mmi @charleseb @motorolamobility @chenyt9

attack11, 2018-10-30:
Where or how did you get the defconfig to build? You follow moto's readme? For wich device are you building?

ezxpro, 2018-10-30:
I'm building for Cedric (Moto g5) and the defcinfig I'm using is the one found in the source code's subdirectory arch/arm/configs/.
The only Readme that comes with the source code, as far as I know, is equal to upstream Linux's Readme, so it tells me nothing very useful.

attack11, 2018-10-30:
You need your device specific defconfig, take a look at this instructions:

https://github.com/MotorolaMobilityLLC/readme/commit/3ac266c5903a5d9cf587838f4c87dbe0e52b01f4#diff-781FE8D88FF07846C93CB02CED5496AE

ezxpro, 2018-10-30:
@attack11 ok, I'll try that asap

ezxpro, 2018-10-31:
@attack11 can't compile using that either. I'll ask for an updated readme file, the you provided is wrong (it links some potter files into the defconfig) and the other one available for Cedric didn't achieve compilation

Jleeblanch, 2018-11-09:
@ezxpro If your running the stock firmware which has the kernel version your trying to build, you can pull the defconfig used from the device itself (located at **/proc/config.gz**).

ezxpro, 2018-11-09:
@Jleeblanch thanks, I managed to pull the defconfig from the location you informed (my device is on stock firmware, which has the same kernel version I intended to compile) and successfully compiled the kernel, but besides the zImage generated after compilation, I get 5 small dtb files (each sized about 220k, whereas the one packed in the boot image has about 1MB). Phone now gets stuck on screen :(

About the readme they posted, that is just a bunch of crap. Not only do they provide broken instructions (e.g a script that has incorrect/broken syntax), but the defconfig generated from Motorola's instructions misses no less than 4000 thousand lines in comparison to the defconfig present in the device itself.

Apart of that, I doubt very much the version of the source code  Motorola released is an exact match of the one they used to build the kernel themselves, I had to fix many wrong/missing header references in the source code before being able to compile the kernel using that defconfig you pointed, which leads me to think they have a significantly different source code at Motorola

attack11, 2018-11-09:
@ezxpro you need to recreate the dtb with the dtb binary

https://github.com/xiaolu/mkbootimg_tools

ezxpro, 2018-11-09:
@attack11 with anykernel I managed to boot the system, but it asked to decrypt data partition (partition was not encrypted before).
I tried then formatting the partition, but upon doing it also gets stuck on bootlogo

ab123321, 2018-11-09:
@ezxpro 

> the defconfig generated from Motorola's instructions misses no less than 4000 lines in comparison to the defconfig present in the device itself.
it's intended, `make oldconfig` automatically generates full defconfig from minified one

attack11, 2018-11-09:
@ezxpro , I saw your sources, Why dont you cloned the repo? You are missing there all commits history,You need to revert this commit:  https://github.com/MotorolaMobilityLLC/kernel-msm/commit/120b471489be6b832ee7a402e0027bb81c9e8c40

ezxpro, 2018-11-09:
@attack11 I work from various computers so I have to clone this repo a lot. That's the reason I didn't clone the entire repo, I just initialized a new repo using the source code provided for my specific device. I'll also take a look into this commit, thanks.

Jleeblanch, 2018-11-09:
@ezxpro As @attack11 mentioned, you need to disable verity on the **/system** and/or **/vendor** partitions before it'll boot. You can revert the commit he pointed out, or simply remove the verify flag in the dtsi file(s) relating to your specific device. Either way will work. 

Also, unless you plan to generate a full boot.img, you don't need to do any more with the dtb files after compiling has completed.

ezxpro, 2018-11-17:
@attack11 @Jleeblanch I successfully compiled it using motorola's instructions by commenting out the gcc-wrapper.py from the makefile AND concatenating the used configs myself (their perls script has broken syntax)
Now I have to see if it boots.

ezxpro, 2018-11-18:
Worked like a charm, now let's customize it :smile:

ezxpro, 2018-11-18:
@ab123321 thanks for pointing that out.
Just  a few remarks, before expanding the defconfig to the full one, it must be renamed to .config and must be within the kernel's root directory. Then you should first issue 
`make defoldconfig`  And just then `make oldconfig` . Issuing make oldconfig directly will generate an entirely new (and empty) defconfig.

closed 2019-11-15 by awheeler-mmi

boris-kolpackov, 2020-07-29:
For anyone else trying to figure our why module signing is enable even though you've explicitly disabled it, CONFIG_SECURITY_LOCKDOWN_LSM=y forces it. So this seems to do the trick:

```
scripts/config --disable SECURITY_LOCKDOWN_LSM
scripts/config --disable MODULE_SIG
```

szylhh, 2023-09-25:
> 对于任何其他试图弄清楚为什么启用模块签名的人，即使您已明确禁用它，CONFIG_SECURITY_LOCKDOWN_LSM=y 也会强制执行它。所以这似乎可以解决问题：
> 
> ```
> scripts/config --disable SECURITY_LOCKDOWN_LSM
> scripts/config --disable MODULE_SIG
> ```
Hello, I also encountered the same problem. How can I solve it? Modified 

`# CONFIG_MODVERSIONS  is not set
# CONFIG_MODULE_SIG is not set
# CONFIG_MODULE_SIG_FORCE is not set`

but it will not take effect after compilation. How should I disable kernel module signing?

szylhh, 2023-09-25:
> 对于任何其他试图弄清楚为什么启用模块签名的人，即使您已明确禁用它，CONFIG_SECURITY_LOCKDOWN_LSM=y 也会强制执行它。所以这似乎可以解决问题：
> 
> ```
> scripts/config --disable SECURITY_LOCKDOWN_LSM
> scripts/config --disable MODULE_SIG
> ```

I want to disable kernel modules signature, but it seems to be forced on. I checked a lot of information and can't turn it off.

tycho-kirchner, 2025-03-19:
I also had to disable several CONFIG_IMA* entries as well, otherwise `MODULE_SIG` would always be re-enabled. Not sure, if all of them are necessary, but I did run:
~~~
./scripts/config --file .config --disable MODULE_SIG -d SECURITY_LOCKDOWN_LSM -d MODULE_SIG_SHA1 \
    --set-val MODULE_SIG_KEY "" -d MODULE_SIG_SHA1 --set-val MODULE_SIG_ALL n -d CONFIG_IMA_KEXEC \
    -d CONFIG_IMA_NG_TEMPLATE -d CONFIG_IMA_DEFAULT_HASH_SHA1 -d CONFIG_IMA_WRITE_POLICY \
    -d CONFIG_IMA_READ_POLICY -d CONFIG_IMA_APPRAISE_BUILD_POLICY -d CONFIG_IMA_APPRAISE_MODSIG \
    -d CONFIG_IMA_KEYRINGS_PERMIT_SIGNED_BY_BUILTIN_OR_SECONDARY -d CONFIG_IMA_DISABLE_HTABLE
~~~