MotorolaMobilityLLC/kernel-msm #193: Kernel source for G4 Plus (athene) OPJ28.108

thejsa, 2018-12-16:
Hey @charleseb and @chenyt9,
Would it be possible to update the kernel source for athene build OPJ28.108?
Thanks in advance.

rajatgupta1998, 2018-12-18:
> Hey @charleseb and @chenyt9,
> Would it be possible to update the kernel source for athene build OPJ28.108?
> Thanks in advance.

That was a soak test build. Haven't seen moto release kernel sources of such builds. I'd rather wait for the public release. Anyways, let's see it is upto them. 👍 👍

tansly, 2019-01-06:
As per GPL, aren't they obliged to release the source if someone (that obtained the build in legal ways) requests it? Even if it's a test build, isn't it distributed to some customers?

charleseb, 2019-01-08:
Hi @rajatgupta1998,

We are working to release this version. It will be available by next week.

Thanks

Jleeblanch, 2019-01-09:
> Hi @rajatgupta1998,
> 
> We are working to release this version. It will be available by next week.
> 
> Thanks

@charleseb I appreciate you're working towards releasing the kernel source for Oreo for athene, but I think we'd be happier to see the kernel source released for the updated build **OPJ28.111-12**. 

Opened request is here: https://github.com/MotorolaMobilityLLC/kernel-msm/issues/196

charleseb, 2019-01-18:
Hi @thejsa,

Could you confirm the version you need? 
I cannot see this version (OPJ28.108) as a released version even as soak test build. 
I'm working to release OPJ28.111-12.

Thanks,
Charles

thejsa, 2019-01-27:
> Hi @thejsa,
> 
> Could you confirm the version you need?
> I cannot see this version (OPJ28.108) as a released version even as soak test build.
> I'm working to release OPJ28.111-12.
> 
> Thanks,
> Charles

Hi Charles, just checked and OPJ28.108 does appear to be the version I have on my phone - I will confess though I sourced it from XDA.

OPJ28.111-12 should be just fine though - thank you!

~Lauren

closed 2019-03-15 by thejsa