MotorolaMobilityLLC/kernel-msm #194: Many bugs during kernel compilation Moto E5

FoxGoddess, 2018-12-23:
I'm trying to compile the latest version of the kernel source from Moto E5 daughter-in-law MMI-OPP27.91-146, I'm having a lot of problems during compilation.

`amaterasu@fox:~/Downloads/msm-MMI-OPP27.91-146$ make -j5
scripts/kconfig/conf  --silentoldconfig Kconfig
drivers/input/touchscreen/nova_mmi/Kconfig:4:warning: ignoring type redefinition of 'TOUCHSCREEN_NT36xxx' from 'tristate' to 'boolean'
  CHK     include/config/kernel.release
  UPD     include/config/kernel.release
  WRAP    arch/arm/include/generated/asm/bitsperlong.h
  WRAP    arch/arm/include/generated/asm/current.h
  WRAP    arch/arm/include/generated/asm/cputime.h
  WRAP    arch/arm/include/generated/asm/early_ioremap.h
  WRAP    arch/arm/include/generated/asm/errno.h
  WRAP    arch/arm/include/generated/asm/emergency-restart.h
  WRAP    arch/arm/include/generated/asm/exec.h
  WRAP    arch/arm/include/generated/asm/hash.h
  WRAP    arch/arm/include/generated/asm/ioctl.h
  WRAP    arch/arm/include/generated/asm/ipcbuf.h
  WRAP    arch/arm/include/generated/asm/kdebug.h
  WRAP    arch/arm/include/generated/asm/local.h
  WRAP    arch/arm/include/generated/asm/irq_regs.h
  WRAP    arch/arm/include/generated/asm/local64.h
  WRAP    arch/arm/include/generated/asm/mcs_spinlock.h
  WRAP    arch/arm/include/generated/asm/msgbuf.h
  WRAP    arch/arm/include/generated/asm/parport.h
  WRAP    arch/arm/include/generated/asm/poll.h
  WRAP    arch/arm/include/generated/asm/param.h
  WRAP    arch/arm/include/generated/asm/resource.h
  WRAP    arch/arm/include/generated/asm/rwsem.h
  WRAP    arch/arm/include/generated/asm/sections.h
  WRAP    arch/arm/include/generated/asm/segment.h
  WRAP    arch/arm/include/generated/asm/scatterlist.h
  WRAP    arch/arm/include/generated/asm/preempt.h
  WRAP    arch/arm/include/generated/asm/sembuf.h
  WRAP    arch/arm/include/generated/asm/serial.h
  WRAP    arch/arm/include/generated/asm/shmbuf.h
  WRAP    arch/arm/include/generated/asm/siginfo.h
  WRAP    arch/arm/include/generated/asm/simd.h
  WRAP    arch/arm/include/generated/asm/sockios.h
  WRAP    arch/arm/include/generated/asm/sizes.h
  WRAP    arch/arm/include/generated/asm/termbits.h
  WRAP    arch/arm/include/generated/asm/socket.h
  WRAP    arch/arm/include/generated/asm/timex.h
  WRAP    arch/arm/include/generated/asm/trace_clock.h
  WRAP    arch/arm/include/generated/asm/unaligned.h
  WRAP    arch/arm/include/generated/asm/termios.h
  CHK     include/generated/uapi/linux/version.h
  UPD     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  UPD     include/generated/utsrelease.h
  HOSTCC  scripts/basic/bin2c
  HOSTCC  scripts/dtc/dtc.o
  HOSTCC  scripts/genksyms/genksyms.o
  CC      scripts/mod/empty.o
  HOSTCC  scripts/selinux/genheaders/genheaders
  HOSTCC  scripts/mod/mk_elfconfig
  HOSTCC  scripts/selinux/mdp/mdp
  HOSTCC  scripts/kallsyms
  HOSTCC  scripts/dtc/flattree.o
  CC      scripts/mod/devicetable-offsets.s
  HOSTCC  scripts/sortextable
  SHIPPED scripts/genksyms/parse.tab.c
  SHIPPED scripts/genksyms/lex.lex.c
  SHIPPED scripts/genksyms/keywords.hash.c
  SHIPPED scripts/genksyms/parse.tab.h
  HOSTCC  scripts/genksyms/parse.tab.o
  MKELF   scripts/mod/elfconfig.h
  GEN     scripts/mod/devicetable-offsets.h
  HOSTCC  scripts/mod/sumversion.o
  HOSTCC  scripts/dtc/fstree.o
  HOSTCC  scripts/genksyms/lex.lex.o
[... 1480 lines cut ...]
 #include <linux/overflow.h>
                            ^
compilation terminated.
scripts/Makefile.build:257: recipe for target 'fs/f2fs/dir.o' failed
make[2]: *** [fs/f2fs/dir.o] Error 1
scripts/Makefile.build:402: recipe for target 'fs/f2fs' failed
make[1]: *** [fs/f2fs] Error 2
make[1]: ** Esperando que outros processos terminem.
  CC      fs/ext4/extents.o
  CC      kernel/time/sched_clock.o
  CC      kernel/time/tick-oneshot.o
  CC      kernel/time/tick-sched.o
  CC      drivers/char/diag/diagchar_hdlc.o
  CC      drivers/char/diag/diagfwd.o
  CC      sound/soc/codecs/audio-ext-clk.o
  CC      sound/soc/codecs/wcd9xxx-resmgr.o
  CC      kernel/time/timer_stats.o
  CC      drivers/char/diag/diagfwd_peripheral.o
  CC      drivers/char/diag/diagfwd_smd.o
  CC      kernel/time/timekeeping_debug.o
  CC      sound/soc/codecs/wcd9xxx-mbhc.o
  BC      kernel/time/timeconst.h
`

linux mint 19.1

Play4NoobWin, 2018-12-23:
I have same error, I did not try, but you can, try adding a generic f2fs support

attack11, 2018-12-23:
You need moto f2fs drivers, https://github.com/MotorolaMobilityLLC/motorola-kernel

FoxGoddess, 2018-12-23:
> You need moto f2fs drivers, https://github.com/MotorolaMobilityLLC/motorola-kernel

I am now having the following errors

`amaterasu@fox:~/Downloads/kernel-msm-MMI-OPP27.91-146$ make -j5
scripts/kconfig/conf  --silentoldconfig Kconfig
drivers/input/touchscreen/nova_mmi/Kconfig:4:warning: ignoring type redefinition of 'TOUCHSCREEN_NT36xxx' from 'tristate' to 'boolean'
  CHK     include/config/kernel.release
  WRAP    arch/arm/include/generated/asm/bitsperlong.h
  WRAP    arch/arm/include/generated/asm/cputime.h
  WRAP    arch/arm/include/generated/asm/current.h
  WRAP    arch/arm/include/generated/asm/early_ioremap.h
  WRAP    arch/arm/include/generated/asm/emergency-restart.h
  WRAP    arch/arm/include/generated/asm/exec.h
  WRAP    arch/arm/include/generated/asm/hash.h
  WRAP    arch/arm/include/generated/asm/errno.h
  WRAP    arch/arm/include/generated/asm/ioctl.h
  WRAP    arch/arm/include/generated/asm/irq_regs.h
  WRAP    arch/arm/include/generated/asm/ipcbuf.h
  WRAP    arch/arm/include/generated/asm/local.h
  WRAP    arch/arm/include/generated/asm/kdebug.h
  WRAP    arch/arm/include/generated/asm/local64.h
  WRAP    arch/arm/include/generated/asm/mcs_spinlock.h
  WRAP    arch/arm/include/generated/asm/msgbuf.h
  WRAP    arch/arm/include/generated/asm/param.h
  WRAP    arch/arm/include/generated/asm/poll.h
  WRAP    arch/arm/include/generated/asm/parport.h
  WRAP    arch/arm/include/generated/asm/preempt.h
  WRAP    arch/arm/include/generated/asm/resource.h
  WRAP    arch/arm/include/generated/asm/sections.h
  WRAP    arch/arm/include/generated/asm/rwsem.h
  WRAP    arch/arm/include/generated/asm/scatterlist.h
  WRAP    arch/arm/include/generated/asm/sembuf.h
  WRAP    arch/arm/include/generated/asm/serial.h
  WRAP    arch/arm/include/generated/asm/shmbuf.h
  WRAP    arch/arm/include/generated/asm/segment.h
  WRAP    arch/arm/include/generated/asm/siginfo.h
  WRAP    arch/arm/include/generated/asm/sizes.h
  WRAP    arch/arm/include/generated/asm/simd.h
  WRAP    arch/arm/include/generated/asm/socket.h
  WRAP    arch/arm/include/generated/asm/sockios.h
  WRAP    arch/arm/include/generated/asm/termbits.h
  WRAP    arch/arm/include/generated/asm/termios.h
  WRAP    arch/arm/include/generated/asm/unaligned.h
  WRAP    arch/arm/include/generated/asm/trace_clock.h
  WRAP    arch/arm/include/generated/asm/timex.h
  CHK     include/generated/uapi/linux/version.h
  UPD     include/generated/uapi/linux/version.h
  HOSTCC  scripts/basic/bin2c
  UPD     include/config/kernel.release
  CHK     include/generated/utsrelease.h
  UPD     include/generated/utsrelease.h
  HOSTCC  scripts/dtc/dtc.o
  CC      scripts/mod/empty.o
  HOSTCC  scripts/genksyms/genksyms.o
  HOSTCC  scripts/mod/mk_elfconfig
  CC      scripts/mod/devicetable-offsets.s
  HOSTCC  scripts/dtc/flattree.o
  SHIPPED scripts/genksyms/parse.tab.c
  HOSTCC  scripts/selinux/genheaders/genheaders
  SHIPPED scripts/genksyms/lex.lex.c
  SHIPPED scripts/genksyms/keywords.hash.c
  SHIPPED scripts/genksyms/parse.tab.h
  HOSTCC  scripts/genksyms/parse.tab.o
  HOSTCC  scripts/selinux/mdp/mdp
  MKELF   scripts/mod/elfconfig.h
  GEN     scripts/mod/devicetable-offsets.h
  HOSTCC  scripts/mod/sumversion.o
  HOSTCC  scripts/dtc/fstree.o
  HOSTCC  scripts/genksyms/lex.lex.o
[... 1562 lines cut ...]
  CC      drivers/misc/qcom/qdsp6v2/q6audio_v2_aio.o
  CC      drivers/misc/modbus_ext.o
  CC      drivers/misc/qcom/qdsp6v2/audio_g711mlaw.o
  CC      drivers/misc/fsusb42_mux.o
  CC      drivers/misc/tps61280.o
  CC      drivers/misc/qcom/qdsp6v2/audio_g711alaw.o
  CC      drivers/misc/qcom/qdsp6v2/audio_mp3.o
  CC      drivers/misc/qcom/qdsp6v2/audio_amrwb.o
  CC      drivers/misc/qcom/qdsp6v2/audio_amrnb.o
  CC      drivers/misc/qcom/qdsp6v2/audio_amrwbplus.o
  CC      drivers/misc/qcom/qdsp6v2/audio_evrc.o
  CC      drivers/misc/qcom/qdsp6v2/audio_qcelp.o
  CC      drivers/misc/qcom/qdsp6v2/amrwb_in.o
  CC      drivers/misc/qcom/qdsp6v2/audio_hwacc_effects.o
  CC      drivers/misc/qcom/qdsp6v2/ultrasound/usf.o
  CC      drivers/misc/qcom/qdsp6v2/ultrasound/usfcdev.o
  CC      drivers/misc/qcom/qdsp6v2/ultrasound/q6usm.o
  LD      drivers/misc/qcom/qdsp6v2/ultrasound/built-in.o
  LD      drivers/misc/qcom/qdsp6v2/built-in.o
  LD      drivers/misc/qcom/built-in.o
  LD      drivers/misc/built-in.o
Makefile:967: recipe for target 'drivers' failed
make: *** [drivers] Error 2
amaterasu@fox:~/Downloads/kernel-msm-MMI-OPP27.91-146$ 
`

FoxGoddess, 2018-12-23:
> I have same error, I did not try, but you can, try adding a generic f2fs support

I am now having the following errors

`amaterasu@fox:~/Downloads/kernel-msm-MMI-OPP27.91-146$ make -j5
scripts/kconfig/conf  --silentoldconfig Kconfig
drivers/input/touchscreen/nova_mmi/Kconfig:4:warning: ignoring type redefinition of 'TOUCHSCREEN_NT36xxx' from 'tristate' to 'boolean'
  CHK     include/config/kernel.release
  WRAP    arch/arm/include/generated/asm/bitsperlong.h
  WRAP    arch/arm/include/generated/asm/cputime.h
  WRAP    arch/arm/include/generated/asm/current.h
  WRAP    arch/arm/include/generated/asm/early_ioremap.h
  WRAP    arch/arm/include/generated/asm/emergency-restart.h
  WRAP    arch/arm/include/generated/asm/exec.h
  WRAP    arch/arm/include/generated/asm/hash.h
  WRAP    arch/arm/include/generated/asm/errno.h
  WRAP    arch/arm/include/generated/asm/ioctl.h
  WRAP    arch/arm/include/generated/asm/irq_regs.h
  WRAP    arch/arm/include/generated/asm/ipcbuf.h
  WRAP    arch/arm/include/generated/asm/local.h
  WRAP    arch/arm/include/generated/asm/kdebug.h
  WRAP    arch/arm/include/generated/asm/local64.h
  WRAP    arch/arm/include/generated/asm/mcs_spinlock.h
  WRAP    arch/arm/include/generated/asm/msgbuf.h
  WRAP    arch/arm/include/generated/asm/param.h
  WRAP    arch/arm/include/generated/asm/poll.h
  WRAP    arch/arm/include/generated/asm/parport.h
  WRAP    arch/arm/include/generated/asm/preempt.h
  WRAP    arch/arm/include/generated/asm/resource.h
  WRAP    arch/arm/include/generated/asm/sections.h
  WRAP    arch/arm/include/generated/asm/rwsem.h
  WRAP    arch/arm/include/generated/asm/scatterlist.h
  WRAP    arch/arm/include/generated/asm/sembuf.h
  WRAP    arch/arm/include/generated/asm/serial.h
  WRAP    arch/arm/include/generated/asm/shmbuf.h
  WRAP    arch/arm/include/generated/asm/segment.h
  WRAP    arch/arm/include/generated/asm/siginfo.h
  WRAP    arch/arm/include/generated/asm/sizes.h
  WRAP    arch/arm/include/generated/asm/simd.h
  WRAP    arch/arm/include/generated/asm/socket.h
  WRAP    arch/arm/include/generated/asm/sockios.h
  WRAP    arch/arm/include/generated/asm/termbits.h
  WRAP    arch/arm/include/generated/asm/termios.h
  WRAP    arch/arm/include/generated/asm/unaligned.h
  WRAP    arch/arm/include/generated/asm/trace_clock.h
  WRAP    arch/arm/include/generated/asm/timex.h
  CHK     include/generated/uapi/linux/version.h
  UPD     include/generated/uapi/linux/version.h
  HOSTCC  scripts/basic/bin2c
  UPD     include/config/kernel.release
  CHK     include/generated/utsrelease.h
  UPD     include/generated/utsrelease.h
  HOSTCC  scripts/dtc/dtc.o
  CC      scripts/mod/empty.o
  HOSTCC  scripts/genksyms/genksyms.o
  HOSTCC  scripts/mod/mk_elfconfig
  CC      scripts/mod/devicetable-offsets.s
  HOSTCC  scripts/dtc/flattree.o
  SHIPPED scripts/genksyms/parse.tab.c
  HOSTCC  scripts/selinux/genheaders/genheaders
  SHIPPED scripts/genksyms/lex.lex.c
  SHIPPED scripts/genksyms/keywords.hash.c
  SHIPPED scripts/genksyms/parse.tab.h
  HOSTCC  scripts/genksyms/parse.tab.o
  HOSTCC  scripts/selinux/mdp/mdp
  MKELF   scripts/mod/elfconfig.h
  GEN     scripts/mod/devicetable-offsets.h
  HOSTCC  scripts/mod/sumversion.o
  HOSTCC  scripts/dtc/fstree.o
  HOSTCC  scripts/genksyms/lex.lex.o
[... 1561 lines cut ...]
  CC      drivers/misc/qcom/qdsp6v2/q6audio_v2.o
  CC      drivers/misc/qcom/qdsp6v2/q6audio_v2_aio.o
  CC      drivers/misc/modbus_ext.o
  CC      drivers/misc/qcom/qdsp6v2/audio_g711mlaw.o
  CC      drivers/misc/fsusb42_mux.o
  CC      drivers/misc/tps61280.o
  CC      drivers/misc/qcom/qdsp6v2/audio_g711alaw.o
  CC      drivers/misc/qcom/qdsp6v2/audio_mp3.o
  CC      drivers/misc/qcom/qdsp6v2/audio_amrwb.o
  CC      drivers/misc/qcom/qdsp6v2/audio_amrnb.o
  CC      drivers/misc/qcom/qdsp6v2/audio_amrwbplus.o
  CC      drivers/misc/qcom/qdsp6v2/audio_evrc.o
  CC      drivers/misc/qcom/qdsp6v2/audio_qcelp.o
  CC      drivers/misc/qcom/qdsp6v2/amrwb_in.o
  CC      drivers/misc/qcom/qdsp6v2/audio_hwacc_effects.o
  CC      drivers/misc/qcom/qdsp6v2/ultrasound/usf.o
  CC      drivers/misc/qcom/qdsp6v2/ultrasound/usfcdev.o
  CC      drivers/misc/qcom/qdsp6v2/ultrasound/q6usm.o
  LD      drivers/misc/qcom/qdsp6v2/ultrasound/built-in.o
  LD      drivers/misc/qcom/qdsp6v2/built-in.o
  LD      drivers/misc/qcom/built-in.o
  LD      drivers/misc/built-in.o
Makefile:967: recipe for target 'drivers' failed
make: *** [drivers] Error 2
amaterasu@fox:~/Downloads/kernel-msm-MMI-OPP27.91-146$ `

attack11, 2018-12-23:
https://github.com/boulzordev/android_kernel_motorola_msm8953/commit/ef57d528200f0882b246989fc538a2f2b9d7485e#diff-5b252455b4f7e91a4ea071001089fd1c

FoxGoddess, 2018-12-23:
> [boulzordev/android_kernel_motorola_msm8953@ef57d52](https://github.com/boulzordev/android_kernel_motorola_msm8953/commit/ef57d528200f0882b246989fc538a2f2b9d7485e)

the same error bro, already tried everything. help

attack11, 2018-12-24:
What commands do you use to build?

FoxGoddess, 2018-12-24:
> You need to specify an "out" folder in new caf kernels
> 
> mkdir -p out
> make O=out clean
> make O=out mrproper
> make O=out <defconfig_name>
> make O=out -j$(nproc --all)
> 
> You extracted defconfig from stock Kernel?

yes, extracted from kernel stock MMI-OPP27.91-146

FoxGoddess, 2018-12-24:
> You need to specify an "out" folder in new caf kernels
> 
> mkdir -p out
> make O=out clean
> make O=out mrproper
> make O=out <defconfig_name>
> make O=out -j$(nproc --all)
> 
> You extracted defconfig from stock Kernel?

same problem bro, not working.

closed 2018-12-29 by FoxGoddess