MotorolaMobilityLLC/kernel-msm #195: Request for OPS28.65-36-6 tag source update for sanders

Dybios, 2019-01-07:
Requesting for an OPS28.65-36-6 tag source update for sanders.

Thanks!

CheckYourScreen, 2019-01-18:
@charleseb any update on this?

charleseb, 2019-01-23:
Hi all,

I'm working to release this version.
It will be available by Tuesday.

Thanks,
Charles

CheckYourScreen, 2019-01-24:
@charleseb thats OPS28.65-36 tag, not OPS28.65-36-6 (december patch).

CheckYourScreen, 2019-02-02:
Hi @charleseb , any eta for OPSS28.65-36-6 tag?

Dybios, 2019-02-02:
> Hi @CheckYourScreen,
> 
> I think you meant OPSS28.65-36-6.
> This version will be available next week.
> 
> Could you reopen this issue, please?
> 
> Thanks,
> Charles

I am not able to reopen this issue again. Will be creating a new one instead.
Thanks.

charleseb, 2019-02-04:
Version released.
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-OPSS28.65-36-6

closed 2019-02-04 by charleseb