MotorolaMobilityLLC/kernel-msm #196: Kernel Source for Moto G4 Plus (athene) OPJ28.111-12

rajatgupta1998, 2019-01-08:
Hi guys, now since we are finally seeing the Oreo 8.1 update (soak) being rolled out to the users, kindly release the linux kernel source as per the GPL. Nice to see Moto G4 Plus get Oreo.

Device: athene
Build Number: OPJ28.111-12

@awheeler-mmi @charleseb @chenyt9 

Hoping to see a reply soon! Thanks.

loonycrd, 2019-01-11:
@awheeler-mmi @charleseb @chenyt9 
Kindly also update the vendor-fci-dtvdriver for digital tv module sources for Moto g4 plus (athene).
Oreo build version: OPJ28.111-12
Nougat build version: NPJS25.93-14-18

charleseb, 2019-01-17:
Hi @rajatgupta1998,

I'm working to release this version.

Thanks,
Charles

bengris32, 2019-01-25:
> Hi @rajatgupta1998,
> 
> I'm working to release this version.
> 
> Thanks,
> Charles

Is it also possible that you can release the camera hal source?

Lyudmila17, 2019-01-26:
> @bengris32, we are working to get the GPL-licensed code released as soon as we can. The camera HAL, however, is proprietary code (for Lenovo as well as some of our partners), so we won't be releasing it. Some of our camera code isn't even open to the majority of the our development staff, it's considered that IP-critical.
> 
> Thanks for your understanding.

Thanks for responding to this, I've asked it on forums once but never got an answer. We can understand you can't release the camera code, but could you release the needed things to make it work with CAF one? Mainly "cam_intf.h" with their structs and everything.

awheeler-mmi, 2019-01-28:
Hey @Lyudmila17, our camera HAL is based on CAF, but we modify the interface quite a bit to get some proprietary features added.  Just looking at cam_intf.h, I see ~58 commits on top of CAF, with about 20% additional content.  Right now this code is very-much proprietary, but I will talk to our camera lead about open-sourcing the interface/structs on new products.

rajatgupta1998, 2019-02-20:
Hi @charleseb @awheeler-mmi . Any update on this? Also we're seeing the OPJ28.111-22 build rolling out to a few people. We're really eager to get the latest kernel source (and dtv, wlan drivers) for Oreo. Please make it possible ASAP. :smile:

amarendra1507, 2019-02-25:
Hi, I have install an OTA which has build OPJ28.108. I want to install OPJ28.111-22, is there any way to do so ? thanks in advance.

batata004, 2019-02-27:
@awheeler-mmi @charleseb please release it!

charleseb, 2019-03-01:
Hi @batata004,

We are working to release this version. It will be released in a few days.

Thanks,
Charles

charleseb, 2019-03-13:
Version released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-OPJ28.111-12

closed 2019-03-13 by charleseb

Lyudmila17, 2019-03-13:
@charleseb what about the commit history?

charleseb, 2019-03-13:
Hi @Lyudmila17,

$> git clone https://github.com/MotorolaMobilityLLC/kernel-msm.git
$> cd kernel-msm
$/kernel-msm> git checkout MMI-OPJ28.111-12
$/kernel-msm> git log

loonycrd, 2019-03-14:
@charleseb could you also update the prebuilt dtbtool binary, the one in readme repository from Nougat build version doesn't seem to work, kernel isn't booting with it.

charleseb, 2019-03-15:
Hi @loonycrd,
Done.
https://github.com/MotorolaMobilityLLC/readme/tree/master/dtbtool_MMI-OPJ28.111-22