MotorolaMobilityLLC/kernel-msm #200: Kernel source release for chef_sprout build PPT29.74-29-2

archie9211, 2019-02-05:
Hi,

Can we have a kernel source release for Motorola one power code named as chef_sprout for PPT29.74-29-2 build please? @charleseb

Thanks.

charleseb, 2019-02-06:
HI @archie9211,

We are working to release this version. It will be released by next week.

Thanks,
Charles

kkmfincorp, 2019-02-14:
Hi @charleseb

Waiting more than a week sir

Thanks.

charleseb, 2019-02-14:
Hi @kkmfincorp,

We are completing the process to this version. I'll release this tag in a few days. 

Thanks,
Charles

kkmfincorp, 2019-02-19:
Hi @charleseb,

Does the process has been finished for chef_sprout build PPT29.74-29-2 #200 .
We are expecting your kind response and release tag.

Thanks,
kkmfincorp

charleseb, 2019-02-20:
Version released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-PPT29.74-29

closed 2019-02-20 by charleseb