MotorolaMobilityLLC/kernel-msm #201: GPL Source Request for Moto G7 Power (ocean) - (PPO29.80-27)

therootlord, 2019-02-08:
Hello, could you provide the corresponding sources ?

@charleseb @chenyt9

Many Thanks!

chirayu19091997, 2019-02-09:
yes if we can have the sources for above tag (PPO29.80-27) for device  that would be great @charleseb @chenyt9 
thanks @therootlord

charleseb, 2019-02-11:
Hi @therootlord and @chirayu19091997,

We are working to release this version. It will be released by next week.

Thanks,
Charles

charleseb, 2019-02-21:
Version released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-PPO29.80-27

closed 2019-02-21 by charleseb

jeraldott, 2019-03-27:
@charleseb Thanks for providing the source code.
Upon review however, I noticed that the kernel source code you provided does not contain the source code for the SAR sensor.

Looking at the phone directly I find 2 drivers related to sar
ocean:/ $ lsmod  | grep sar
abov_sar               40960  0
sx932x_sar             40960  0
sensors_class          20480  2 abov_sar,sx932x_sar
ocean:/ $

Inside the source code, the dtsi only points to abov_sar
~/kernel-msm-MMI-PPO29.80-27/arch/arm64/boot/dts/qcom$ rgrep 'sar' ./
./sdm450-ali-common-overlay.dtsi:               cyttsp,sar-status-reg = <0x50>;
./sdm632-channel-common-overlay.dtsi:           compatible = "abov,abov_sar";
./sdm632-channel-na-dvt1b-overlay.dts:          compatible = "abov,abov_sar";
./sdm632-ocean-common-overlay.dtsi:             compatible = "abov,abov_sar";

Looking back at the phone I see it is shown to match up with the dtsi file.
1|ocean:/sys/bus/i2c/drivers/abov_sar/2-0020 $ ls -la
ls: ./calibrate: Permission denied
ls: ./modalias: Permission denied
ls: ./uevent: Permission denied
ls: ./name: Permission denied
total 0
drwxr-xr-x 3 root root 0 1969-12-31 19:00 .
drwxr-xr-x 7 root root 0 1969-12-31 19:00 ..
lrwxrwxrwx 1 root root 0 1969-12-31 19:00 driver -> ../../../../../../bus/i2c/drivers/abov_sar
lrwxrwxrwx 1 root root 0 1969-12-31 19:00 of_node -> ../../../../../../firmware/devicetree/base/soc/i2c@78b6000/abova96t316@20
drwxr-xr-x 2 root root 0 1969-12-31 19:00 power
lrwxrwxrwx 1 root root 0 1969-12-31 19:00 subsystem -> ../../../../../../bus/i2c

 I assume the driver was created as a kernel driver and therefore the source code should be provided, correct? As this is also related to SAR regulations, this driver is a necessity to ensure proper RF power under FCC regulations.

Thanks.
-Jerald

charleseb, 2019-03-28:
Hi @jeraldott,

You can find the files here:
https://github.com/MotorolaMobilityLLC/motorola-kernel-modules/releases/tag/MMI-PPO29.80-27
I've missed to include this project, sorry about that.

Thanks,
Charles

jeraldott, 2019-03-28:
Hi @charleseb,
Thanks! No problem.
-jerald

no-identd, 2019-04-15:
Please, can you also release:

PPO29.71-34
PPO29.80-14 
PPO29.114-30

?

no-identd, 2019-08-06:
Adding PPO29.114-134 to the list of requested source code releases.