MotorolaMobilityLLC/kernel-msm #202: Moto G7 Plus (lake) - Kernel source for tag PPW29.98-28

Jleeblanch, 2019-02-08:
Hey, @charleseb, @chenyt9, @awheeler-mmi 

When you guys get a chance, can you release the kernel source for tag **PPW29.98-28** for the Moto G7 Plus (lake). 

Thank you,
Joshua.

charleseb, 2019-02-21:
HI @Jleeblanch,

We are working to release this version. It will be available by next week.

Thanks,
Charles

Jleeblanch, 2019-03-19:
@charleseb just checking in on how things are looking.

charleseb, 2019-03-19:
Hi @Jleeblanch,

Sorry about the delay. It was needed to get some processes concluded before to release. I'm going to push this today.

Thanks,
Charles

Jleeblanch, 2019-03-19:
No worries @charleseb, 

Thank you for the heads up

charleseb, 2019-03-19:
Version released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-PPW29.98-28

closed 2019-03-19 by charleseb