MotorolaMobilityLLC/kernel-msm #204: Request for kernel source of chef_sprout PPTS29.74-41 build.

smyaknti, 2019-03-02:
Hi,

Can we have a kernel source release for Motorola one power code named as chef_sprout for PPTS29.74-41 build please? @charleseb

Thanks.

archie9211, 2019-03-29:
@charleseb can we have an update please. 
or if there are no changes in kernel source in new update, then please let us know.

charleseb, 2019-03-31:
Hi @smyaknti and @archie9211,

We are working to release this version. It will be available in a few days.

Thanks,
Charles

Anush12133, 2019-04-18:
Hi,

Can we have a kernel source release for Motorola one power code named as chef_sprout for PPT29.74-41-2 build please? @charleseb . Also one more request there is an big issue of incompatibility of ims in source please do make it compatible for better usability or if u have other solution for that also will be fine @charleseb 

Thanks.

awheeler-mmi, 2019-04-19:
Hey @Anush12133,

What do you mean by IMS incompatibility? Our IMS stack is closed-source, but I would not expect a recompile of the kernel to  break compatibility with the IMS stack. Let me know what you're looking for.

Thanks,

-andrew

Anush12133, 2019-04-19:
@awheeler-mmi I didn't mean kernel source broke ims stack or something ...

Anush12133, 2019-04-19:
I just asked when is PPT29.74-41-2 kernel Source update releasing. Ims was a different question

Anush12133, 2019-04-19:
Because I did require ims stuff which Chef uses

awheeler-mmi, 2019-04-19:
Gotcha.  Yeah, @charleseb is working on the kernel release so we should release it soon.  Note it's a holiday here, so give us until next week.  

Many of our newer devices use a third-party IMS stack, instead of the QCOM stack, so that may explain the differences you are noticing.

Anush12133, 2019-04-19:
I would wish moto to make ims stack open too as it's becoming difficult for us adapt things for making VOLTE properly function

luk1337, 2019-04-19:
That's interesting, do you have qcom devices using third party IMS implementations as well?

awheeler-mmi, 2019-04-19:
@Anush12133 Gotcha, and, yeah, that would be tricky.  Our IMS stacks are third-party and our licenses very much restrict our ability to share the code.  We don't even have access to all the source.

One major AOSP gap is IMS, and it forces all manufacturers to either use third-party solutions, or roll their own.  It would be great if Google open-sourced a production-quality stack.  I suspect that if they were to do that then manufacturers would adopt it.

Anush12133, 2019-04-19:
Are u using CAF IMS stack as CAF dailer and CAF  device/sepolicy did made a change and VOLTE enabled partially

Anush12133, 2019-04-19:
I was able to receive calls and send SMS on Volte

awheeler-mmi, 2019-04-19:
After some looking, I believe that our QCOM products are still using the IMS framework, but we call out to 3rd parties for some IMS features like RCS, etc.  I am not sure how much our IMS stack depends on these non-QC 3rd party components for basic functionality.  But QCOM releases their IMS framework to us as mostly pre-built, proprietary binaries.  Which CAF IMS components are you referring to?

Anush12133, 2019-04-19:
I just reffered the ims part which is integrated in dialer app

Anush12133, 2019-04-19:
And the modem Chef_sprout uses is a M636 which is specific modem by Qcomm for SDM636

awheeler-mmi, 2019-04-19:
Oh, gotcha.  Yeah, the dialer for your build looks to be "ours", but it's not open-sourced.  (For other builds we use the GMS Dialer, which closed-sourced to us).  I will let our Dialer team lead know that customers are interested in us open-sourcing it!

Anush12133, 2019-04-19:
I will be happy to see  Motorola making it open source . It will help devs like us a lot .. Hoping it will be soon :) . Thanks in advance Motorola :)

Anush12133, 2019-04-19:
No the dailer I used was directly from CAF

awheeler-mmi, 2019-04-19:
Oops, the Dialer lead just let me know that all Android One targets are using the GMS dialer, which is closed-source.

Anush12133, 2019-04-19:
Can any of your team member guide us through it for fixing it would be great for us :).. We have proper logs for it too

awheeler-mmi, 2019-04-19:
No, sorry.  We are happy to see customers hack our devices, but we can't do much to assist with it -- particularly in cases where we have disabled functionality due to carrier requirements.

Anush12133, 2019-04-19:
Currently I m referring CAF Dailer commits also device/sepolicy commits of CAF

Anush12133, 2019-04-19:
It's OK. But just hints will do

awheeler-mmi, 2019-04-19:
We don't use the CAF dialer for any builds, so I'm not sure where to point you.

Anush12133, 2019-04-19:
I will try to explain situation so that you can try pointing .What I did is I used dailer and device/sepolicy from CAF. Which with stock ims.apk gave me partial VOLTE (I can send SMS and I am able to receive calls on VOLTE) But unable to call nor Receive SMS(SMS receives but very rarely) . IMS gets registered for some cases asper logs and after that simply gets to Unregistered state

awheeler-mmi, 2019-04-20:
Gotcha.  I am surprised this combination is working at all.  Most of these features are controlled via the framework, not so much the app. Maybe the kind folks at XDA could help.

Hasaber8, 2019-04-22:
> Gotcha. I am surprised this combination is working at all. Most of these features are controlled via the framework, not so much the app. Maybe the kind folks at XDA could help.

Just give us some fix to get volte working.

awheeler-mmi, 2019-04-22:
@Hasaber8 we definitely don't have any fix for the dialer/ims combination mentioned.  But before enabling volte for a specific carrier we need to comply with a number of carrier requirements, including extensive testing.  We can't release hacks to essentially bypass the carrier compliance requirements.

Hasaber8, 2019-04-22:
> @Hasaber8 we definitely don't have any fix for the dialer/ims combination mentioned. But before enabling volte for a specific carrier we need to comply with a number of carrier requirements, [...]

You cant but you can guide us to do it.

awheeler-mmi, 2019-04-22:
If we had guidance and we were able to share it, we would!  We do help with build/boot issues, and are happy to see folks hacking our products (for good, not black-hat stuff).  But we really are not in a good position to help bypass carrier requirements.

Hasaber8, 2019-04-22:
Not just one carrier, all infact. Allmost 80% carrier have volte, just saying you can help us with the apk and stuff, not just configs

awheeler-mmi, 2019-04-22:
But if you install/use GMS dialer then VoLTE works, right?

Hasaber8, 2019-04-22:
> But if you install/use GMS dialer then VoLTE works, right?

NA, not fully, just sms, 
We want to get working fully

smyaknti, 2019-04-22:
> Not just one carrier, all infact. Allmost 80% carrier have volte, just saying you can help us with the apk and stuff, not just configs

Adding to the stuff he said. IMS is registered and incoming calls give the ringing tone for the caller, but not for the receiver, the device just stays there with no response. Hence I suspect the carrier side is fine, we are not able to get IMS calls working on the device end. As stock has VoLTE calls working fine, we are asking for any specific patches if they are being used which we are missing. 

Thanks

awheeler-mmi, 2019-04-22:
Gotcha, but the stock build uses the GMS dialer.  Perhaps you could try the Moto dialer from a chef (retail) build, rather than the GMS dialer from a chef_sprout (Android One) build.

Hasaber8, 2019-04-22:
Hmmm, we will try the dialerbut i still feel its the problem with ims.apk, but we will keep you updated.

smyaknti, 2019-04-22:
> Gotcha, but the stock build uses the GMS dialer. Perhaps you could try the Moto dialer from a chef (retail) build, rather than the GMS dialer from a chef_sprout (Android One) build.

Okay, we shall try a full set of stock apps and see. We may have missed some important apps last time because the baseband went unknown and SIM cards weren't detected. Hopefully we dont mess up this time. Any precautions will be helpful.

Hasaber8, 2019-04-22:
> Gotcha, but the stock build uses the GMS dialer. Perhaps you could try the Moto dialer from a chef (retail) build, rather than the GMS dialer from a chef_sprout (Android One) build.

If you dont mind could you look in our commits once and check if updated blobs properly?

awheeler-mmi, 2019-04-22:
Where are your commits?

Hasaber8, 2019-04-22:
> Where are your commits?
pie-dev brach for device-tree and vendor-tree
https://github.com/Chef-Sprout/android_vendor_motorola_chef/tree/pie-dev

https://github.com/Chef-Sprout/android_device_motorola_chef/tree/pie-dev
lineage-16.0 for kernel
https://github.com/Chef-Sprout/android_kernel_motorola_chef

awheeler-mmi, 2019-04-22:
Also, maybe try the AOSP dialer.  Our dialer is closer to AOSP than CAF.

Hasaber8, 2019-04-22:
> Also, maybe try the AOSP dialer. Our dialer is closer to AOSP than CAF.

We will just check our commits once and tell what are we missing.

awheeler-mmi, 2019-04-22:
Hmm... I looked, and it's just a bunch of binaries to me.  I'll ask the Telephony guys to take a look.

Hasaber8, 2019-04-22:
> Hmm... I looked, and it's just a bunch of binaries to me. I'll ask the Telephony guys to take a look.

Thankyou!

charleseb, 2019-05-02:
Version released: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-PPTS29.74-41-2

closed 2019-05-02 by charleseb