MotorolaMobilityLLC/kernel-msm #205: build 64Bit kernel

amaterasushen, 2019-03-08:
I have Motorola Nora device, and I wonder if I can compile the kernel in 64bit and make the kernel work 100% in 64bit. The default system runs in 32bit.

mvaisakh, 2019-03-21:
You can if the board supports it. Moto G5 (Cedric) and also the plus variant (Potter) both have 32bit (arm) variants, but now run 64bit (arm64). You can use kernel sources linked in the xda forums to have a reference. And yes it is pretty bugless (unnoticeable bugs.

amaterasushen, 2019-03-25:
> You can if the board supports it. Moto G5 (Cedric) and also the plus variant (Potter) both have 32bit (arm) variants, but now run 64bit (arm64). You can use kernel sources linked in the xda forums to have a reference. And yes it is pretty bugless (unnoticeable bugs).

I use a motorola e5, in xda there is no malaria talking about 64bit. Can you tell me why some motorola devices run over 32bits and some over 64bits?

mvaisakh, 2019-03-26:
Older Motorola devices run only on 32bit (I honestly don't know why), the newer devices, starting from Moto X4, are all 64bit.

closed 2019-06-17 by amaterasushen