MotorolaMobilityLLC/kernel-msm #207: Moto G4 Plus (athene) - Kernel source OPJ28.111-22

loonycrd, 2019-03-31:
Hey @awheeler-mmi, @charleseb, and @chenyt9,

Can you update the kernel source tag for current OPJ28.111-22 official release as well as dtv module sources in vendor-fci-dtvdriver repo for the Moto G4 Plus (athene)?

Thank you,
Rohit

charleseb, 2019-03-31:
Hi @loonycrd,

We are working to release this version for both projects.

Thanks,
Charles

charleseb, 2019-04-01:
Hi @loonycrd,

You can find vendor-fci-dtvdriver artifacts here: https://github.com/MotorolaMobilityLLC/vendor-fci-dtvdriver/releases/tag/MMI-OPJ28.111-12

Thanks,
Charles

charleseb, 2019-04-01:
We are working to release the version OPJ28.111-22. It will be released in a few days.

Thanks,
Charles

loonycrd, 2019-04-10:
@charleseb any update on this?

charleseb, 2019-04-12:
Version released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-OPJ28.111-22

closed 2019-04-12 by charleseb