MotorolaMobilityLLC/kernel-msm #209: Moto G6 (ali) - Kernel source for tag PPS29.55-24

Jleeblanch, 2019-04-02:
Hey @charleseb 

When you get a chance, can you release the kernel source for tag PPS29.55-24 for the Moto G6 (ali)?

Thank you, 
Joshua

charleseb, 2019-04-02:
HI @Jleeblanch,

We are working to release this version. It will be released by next week.

Thanks,
Charles

Jleeblanch, 2019-04-22:
@charleseb any updates on the kernel source?

awheeler-mmi, 2019-04-24:
Hi @Jleeblanch, @charleseb is on this.  He's on holiday right now, but will get to them released as soon as he is back, next week.

Jleeblanch, 2019-04-25:
@awheeler-mmi, ok no worries. Thank you for the heads up

wh0dat, 2019-04-28:
@awheeler-mmi @charleseb , any chance yall can release source for stock timeweather widget? I'd like to get this working in custom ROM on the lockscreen.
Thanks in advance!! & Sorry to ask here

charleseb, 2019-05-01:
Version released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-PPS29.55-24

closed 2019-05-01 by charleseb