MotorolaMobilityLLC/kernel-msm #210: RHannah kernel source

wh0dat, 2019-04-09:
Hi, can Motorola please provide a working kernel for Rhannah? The one here did not boot for Rhannah , however it booted fine for hannah_t and ahannah.

Thank you in advance!
~Barry Allen
<flash@remainsilent.net>

GodcraftPlata, 2019-04-09:
Hi there! Can Motorola publish a Kernel for Rhannah? (Obviously working) The one provided didn't boot on Rhannah, but it worked fine on other variants (like hannah_t and ahannah).

Thanks in advance for your help.
By: GodcraftPlata.

charleseb, 2019-04-30:
Hi @GodcraftPlata,

Could you describe the version tag, please?

Thanks,
Charles

wh0dat, 2019-04-30:
> Hi @GodcraftPlata,
> 
> Could you describe the version tag, please?
> 
> Thanks,
> Charles

No longer needed , thanks anyway Charles.

closed 2019-04-30 by charleseb