MotorolaMobilityLLC/kernel-msm #211: Motorola One (deen) - Audio Issue on Custom Rom

100Daisy, 2019-04-24:
Rom: RessurectionRemix v7.0.1
Device tree: www.github.com/100Daisy/android_device_motorola_deen
Stock Vendor Tree

LOG:
W APM_AudioPolicyManager: getOutputForDevice() could not find output for stream 1, sampling rate 48000, format 0x1, channels 0x3, flags 0x4
E AudioFlinger: createTrack() getOutputForAttr() return error -38 or invalid output handle
E IAudioFlinger: createTrack returned error -38
E AudioTrack: AudioFlinger could not create track, status: -38 output 548377968

luk1337, 2019-04-24:
...why do you think they care?

100Daisy, 2019-04-24:
> ...why do you think they care?

Maybe they not, someone other can help me

alexa-v2, 2019-08-11:
Not sure this is correct place for this

Contact your maintainer

closed 2019-08-11 by 100Daisy