MotorolaMobilityLLC/kernel-msm #216: Motorola One Vision (kane) - Kernel source for tag PSA29.97-13-9

GtrCraft, 2019-05-16:
Hi,

Can you please release kernel source of Motorola One Vision tag PSA29.97-13-9?

awheeler-mmi, 2019-05-16:
Hey @GtrCraft, the Moto One Vision is a Samsung Exynos-based device, so we're going to create "kernel-slsi" and probably a few other new projects for this release.  I'll leave this issue open until we get the projects created, and then will re-open your request under kernel-slsi.

Give us a bit of time for this release.  We should get the new projects (at least kernel-slsi) created today, but it will likely take some additional time to get make our first SLSI release.

awheeler-mmi, 2019-05-16:
Moving to https://github.com/MotorolaMobilityLLC/kernel-slsi/issues/1 .

closed 2019-05-16 by awheeler-mmi