MotorolaMobilityLLC/kernel-msm #220: Kernel Nash PPX29.159-10

Uzephi, 2019-06-05:
Since it is rolling out, can we get Kernel sources for Nash. Build PPX29.159-10?

jbats, 2019-06-08:
Still waiting on a response for this.

npjohnson, 2019-06-12:
@charleseb @chenyt9 
Hate to tag, but you all were very helpful in attaining source when Oreo dropped, just thought I'd look y'all in on the discussion (:

Uzephi, 2019-06-25:
I assume it is because they haven't finalized the kernel. We got a -19 build distributed recently. If you are fixing the kernel. Please fix the way your kernel handles rendering. com.android.test.uibench program looks horrid currently on any release of the Z2F. Below is the results on pie. It should be consistently below 0.5. I'll wait patiently if it would assist in making the kernel release better.

https://cdn.discordapp.com/attachments/397594172149727234/592744548174659586/Screenshot_20190624-125444.png

Uzephi, 2019-07-10:
PPX29.159-19 officially went out as an OTA per our friends in Brazil. Can we please be compliant with this release since it is public and follow GPL?

Uzephi, 2019-07-10:
@charleseb @chenyt9 could we get a monthly update on this request now that Pie is rolling out officially and publicly?

awheeler-mmi, 2019-07-10:
Hey @Uzephi and others, @charleseb is on vacation for the next two weeks.  In the meantime we are trying to figure out how to get this release out, given the interest in a nash P kernel. I'll keep y'all posted!

Uzephi, 2019-07-31:
Any update on this? PPX-29.159-19 released in Brazil and India. We have firmware as well via LMSA tool. Can we please get the kernel sources? 

License reminder
https://www.gnu.org/licenses/gpl-violation.en.html

charleseb, 2019-07-31:
Hi @Uzephi,

I'm working to release this. This version will be available by the next week.

Thanks,
Charles