MotorolaMobilityLLC/kernel-msm #223: VOLTE ISSUE ON CHEF IN CUSTOM ROMS

Anush12133, 2019-06-21:
Hi,

Can we get new ims.apk and qti-telephony which is compatible with custom roms

i have logs for it too 

[BCC.txt](https://github.com/MotorolaMobilityLLC/kernel-msm/files/3314407/BCC.txt)

its being a long time we are facing such issues please do check this issue

walkerobCMT, 2019-06-21:
Hey @Anush12133, we don't currently release pre-built APKs or most of the qti-telephony components as part of our open-source releases.   I understand the request, though, and will take it up with the executive staff here in Chicago.

awheeler-mmi, 2019-06-21:
Hey @Anush12133, we don't currently release pre-built APKs or most of the qti-telephony components as part of our open-source releases. I understand the request, though, and will take it up with the executive staff here in Chicago.

Anush12133, 2019-06-21:
Hoping for quick solution from your team ...Thanx Moto ..Hope you help the community for healthy development :)

SyberHexen, 2019-08-08:
> Hey @Anush12133, we don't currently release pre-built APKs or most of the qti-telephony components as part of our open-source releases. I understand the request, though, and will take it up with the executive staff here in Chicago.

For reference this is also an issue with PIE variants on devices such as Payton, Evert, Beckham.

An AOSP compatible implementation to QTI-telephony-common.jar and ims.apk would greatly be appreciated by the community!

awheeler-mmi, 2020-12-22:
I'm closing this issue for now, since it is quite old.  I have spoken with the Telephony teams here, but I don't expect any changes in the short-term.  Sorry for the bad news.

closed 2020-12-22 by awheeler-mmi