MotorolaMobilityLLC/kernel-msm #231: GPL Source Request for Moto G7 Power (ocean) - (PPO29.71-34, PPO29.80-14, PPO29.114-30))

no-identd, 2019-08-06:
Please provide the source code for the following tags:

PPO29.71-34
PPO29.80-14
PPO29.114-30

Thanks.

rottenwheel, 2019-09-01:
@charleseb Hi there. In the past you helped the community get kernel sources for another Motorola device, could you help us get sources for ocean as well please? XDA developers will appreciate it!

holyangel, 2019-09-02:
Could we possibly get source with the extra modules already included and build-able please?  

Anyone should be able to sync the release you post and build it and everything just work. With the last ocean source release touchscreen and some other things don't work and converting the extra modules zip you released is a giant pain when building the kernel outside of rom source.

no-identd, 2019-09-02:
+1 to what @holyangel said. Also cf. #201. Cc @chenyt9 @awheeler-mmi.

…and given the rather very slow handling by Motorola (not just for this device, but in general), I figure I might as well also cc @bkuhn.

charleseb, 2019-09-02:
Hi all,

I'm starting the process to push these tags.

Thanks,
Charles

awheeler-mmi, 2019-09-03:
@holyangel do you mean motorola-kernel-modules? That project is separate here at Motorola, so we release it separate on Github, as well, so that y'all get the commit history.  Many of our kernel drivers are outside the kernel tree due to Android's Project Treble requirements.  We could potentially drop the release as a single zipfile, much like other manufacturers, but at the cost of the commit history.

holyangel, 2019-09-03:
@awheeler-mmi yes that is the project I was referring to.. but no I wouldn't need a separate zip of them. I do appreciate you guys keeping the commit history. I was just hoping they could be included inside the kernel source on a release, so we don't have to do a bunch of manual editing to build them with the kernel in one go, stand-alone instead of in rom source/separate. If it can't be done that's fine though.

awheeler-mmi, 2019-09-03:
@holyangel Got it.  That could make the entire package more easily buildable, and easier for us to produce the build instructions.  We'll look into it.

charleseb, 2019-09-09:
Version released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-PPO29.114-30

no-identd, 2019-09-10:
Thanks, one down, two to go! Or rather, three, as by now I have one more request for this device, namely, for:

PPO29.114-134

charleseb, 2019-10-03:
Hi @no-identd,

This version is in our pipeline. It will be released in two weeks.

Thanks,
Charles

closed 2019-11-05 by charleseb