MotorolaMobilityLLC/kernel-msm #235: GPL Source Request for Moto G7 (river) + Power (ocean) - (PPO29.114-134-2)

wh0dat, 2019-09-25:
Please include techpack also

aeqrs25, 2019-09-25:
Release the kernel Motorola please.

wh0dat, 2019-10-12:
On River ... In stock , if listening to BT and get incoming call and accept it , BT completely dies after the call and cannot listen to music unless reboot. @charleseb @awheeler-mmi please give techpack for river?? I was planning to buy a ocean but been waiting for techpack for it aswell. PLEASE??!
Thanks in advance

~Barry Allen

electimon, 2019-10-12:
Please we need techpack

Geofferey, 2019-10-12:
Hi, the techpack is one of the last things we need for functional audio on our custom roms. It would be very much appreciated if you could also include that source as well. Thank you for your consideration.

charleseb, 2019-10-30:
Version released: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-PPO29.114-134
techpack: https://github.com/MotorolaMobilityLLC/vendor-qcom-opensource-audio-kernel/releases/tag/MMI-PPO29.114-134

motog7, 2019-10-31:
Thanks Charles , this is only for ocean? What about river?

charleseb, 2019-10-31:
Hi @motog7,

Working on it. It will be available by next week.

Thanks,
Charles

rajatgupta1998, 2019-11-02:
> Hi @motog7,
> 
> Working on it. It will be available by next week.
> 
> Thanks,
> Charles

Hi @charleseb . Please consider releasing the kernel and the techpack sources for the build PPOS29.114-134-4 for Moto G7 (River) if possible, as it is the most recent one. :)

charleseb, 2019-11-05:
Versions released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-PPOS29.114-134-2
https://github.com/MotorolaMobilityLLC/vendor-qcom-opensource-audio-kernel/releases/tag/MMI-PPOS29.114-134-2
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-PPOS29.114-134-2_RIVER
https://github.com/MotorolaMobilityLLC/vendor-qcom-opensource-audio-kernel/releases/tag/MMI-PPOS29.114-134-2_RIVER

closed 2019-11-05 by charleseb

wh0dat, 2019-11-05:
thanks @charleseb