MotorolaMobilityLLC/kernel-msm #236: GPL Source Request for Moto E5 (codename Nora) (OPPS27.91-176-11-5)

IMPERADOR-MENDIGO, 2019-10-04:
please release the recent version of the Nora kernel OPPS27.91-176-11-5.

thanks.

Play4NoobWin, 2019-10-05:
Soon, maybe

amaterasushen, 2019-10-07:
I want it too.

charleseb, 2019-10-11:
Hi @IMPERADOR-MENDIGO, @Play4NoobWin, @amaterasushen,

We are starting the process to release this version. It will be available in two weeks.

Thanks,
Charles

charleseb, 2019-11-05:
Hi @IMPERADOR-MENDIGO,

I'm still working on this. It will be released this week. Sorry about the delay.

Thanks,
Charles

charleseb, 2019-11-13:
Version released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-OPPS27.91-176-11-5

closed 2019-11-13 by charleseb

amaterasushen, 2019-11-13:
> Version released:
> https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-OPPS27.91-176-11-5

it didn't take long hahahahah. Thanks for releasing.