MotorolaMobilityLLC/kernel-msm #240: motorola e5 MMI-OPP28.44-26-17

amaterasushen, 2019-12-03:
Two days ago a version of moto e5 (codename nora) came out MMI-OPP28.44-26-17 in the repository saying it was version 8.1.0 of motorola Nora. however it is 8.0.0. Moto e5 and Moto e5 plus will be upgraded to Android 8.1 How was the moto e5 play(james)? or just the kernel version that was updated for android 8.1?

awheeler-mmi, 2019-12-03:
Hey @amaterasushen, all OPP28.* builds should be 8.1.  I just checked the build properties, and confirmed this.  Where are you getting that it's 8.0?

(Moto ancient history: The "28" in our build number is locked the Android SDK version +1.  It is meant to be the "Motorola" version number, and is SDK+1 for reasons lost to time.  Perhaps because we are always one step ahead of Android!!!)

amaterasushen, 2019-12-03:
> Hey @amaterasushen, all OPP28.* builds should be 8.1. I just checked the build properties, and confirmed this. Where are you getting that it's 8.0?
>
> (Moto ancient history: The "28" in our build [...]

Hello friend, I did not quite understand what you wrote.
all versions of the nora kernel come out in version 27 *, the last released version was: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-OPPS27.91-176-11-5 with the following description "Nora push for Oreo 8.0.0", and version MMI-OPP28.44-26-17 had the following description "nora push for oreo 8.1.0". I would like to know why you had this change.

It made me think that the moto e5 and moto e5 plus would be upgraded to android 8.1.

awheeler-mmi, 2019-12-03:
OK, I just looked into this. The e5 Tracfone version is a slightly different product.  It's using an MSM8920, and launched a few months later than the 8917-based e5.

I believe we utilized 8920 in North America due to some NA modem requirements that are not met with 8917.  And since it launched later, we released it with 8.1.

I don't expect the 8917-based e5 to get an 8.1 update.

amaterasushen, 2019-12-04:
> OK, eu apenas olhei para isso. A versão do e5 Tracfone é um produto ligeiramente diferente. Ele está usando um MSM8920 e foi lançado alguns meses depois do e5 baseado em 8917.
> 
> Acredito que utilizamos 8920 na América do Norte devido a alguns requisitos de modem de NA que não são atendidos com 8917. E desde que foi lançado mais tarde, o lançamos com 8.1.
> 
> Não espero que o e5 baseado em 8917 obtenha uma atualização 8.1.

thanks for the clarification.

closed 2019-12-04 by amaterasushen