MotorolaMobilityLLC/kernel-msm #242: Source request for PPO29.80-61

asprakash, 2019-12-12:
Hi. I request the kernel source for MOTO G7, PPO29.80-61, Model : XT1962-6

TIA

charleseb, 2019-12-12:
Hi @asprakash,

We are working to release this version. It will be released by next week.

Thanks,
Charles

asprakash, 2019-12-13:
Hi Charles. Thanks for the prompt reply and considering my request. Looking forward your further update.

May I know what are differences b/w these various kernel version though the Moto G7 Play/Power are having the same processor?. I found a nearest version MMI-PPO29.80-27. Is this different from my requested version PPO29.80-61?.

TIA.

asprakash, 2019-12-23:
Hi. Any updates on the kernel source?

charleseb, 2019-12-23:
Hi @asprakash,

Sorry about the delay. I'm still working to process this version before to push.
We'll release both versions MMI-PPO29.80-27 and MMI-PPO29.80-61.
These versions will be released by next week.

Thanks,
Charles

asprakash, 2020-01-03:
@charleseb Still waiting for your release.

charleseb, 2020-01-07:
Version released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-PPO29.80-61

asprakash, 2020-01-08:
Thanks @charleseb 
It will be very helpful if there is any document/steps to compile this source.
TIA

charleseb, 2020-01-17:
Version released: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-PPO29.80-27-RIVER
Readme: https://github.com/MotorolaMobilityLLC/readme/blob/master/MMI-PPO29.80-27_RIVER.txt

closed 2020-01-17 by charleseb