MotorolaMobilityLLC/kernel-msm #243: Moto One Power (Chef) Kernel source request QPT30.61-18

Hasaber8, 2019-12-16:
Hey , with Stable Q rolling out for Chef, I request you to release the kernel source for it, version QPT30.61-18 has a kernel build date of 3rd December.
It will be nice of you all to release it as soon as possible.

Rohan.

archie9211, 2019-12-22:
Any updates? ETA?

therootlord, 2019-12-26:
Up @charleseb

Hasaber8, 2020-01-08:
@charleseb any update?

charleseb, 2020-01-08:
Hi all,

I'm starting the process to release this version.
It will be released by next week.

Thanks,
Charles

Hasaber8, 2020-01-15:
1/x weeks down.

Hasaber8, 2020-01-21:
> Hi all,
> 
> I'm starting the process to release this version.
> It will be released by next week.
> 
> Thanks,
> Charles

2/x weeks down.

charleseb, 2020-01-21:
Version released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-QPT30.61-18

closed 2020-01-21 by charleseb