MotorolaMobilityLLC/kernel-msm #246: [REQUEST] Kernel source release for Moto g8+ (doha)

mauro235, 2020-01-21:


charleseb, 2020-01-23:
Hi @mauro235,

Could you describe the tag version you need, please?

Thanks,
Charles

mauro235, 2020-01-23:
Thanks for your fast response @charleseb !

I'm updating the comment with the tag version which is PPIS29.65-21-2

charleseb, 2020-01-23:
Hi @mauro235,

I'm starting the process to release this version.

Thanks,
Charles

borislavba, 2020-01-31:
@mauro235, I've just "received" version PPIS29.65-24-2. According to the changelog the android version is update to the 1 November 2019 Security Patch. 
Isn't it better to update the request to the latest one or open a second issue for it?

N. B. I'm new around these requests and stuff, sorry if I'm asking obvious things :)

mauro235, 2020-01-31:
That's the latest one from Motorola, so, how do I request something that doesn't yey?

borislavba, 2020-01-31:
Let's say that my intention was to notify you that there is an official newer version and if we should wait a bit to see if it's better to request that instead. Maybe @charleseb can tell us if there are important changes in the kernel so that we should request PPIS29.65-24-2?

mauro235, 2020-01-31:
Seems that it's virtually the same as PPIS29.65-21-2 (which is also November 2019 patch), but a build for India which came later that's why a higher number.

borislavba, 2020-01-31:
OK, thanks for the info. Actually I'm in Bulgaria and the "channel" is reteu, so this version might be available in Europe too. I'll try to use (or to build myself if I have the time) a custom kernel based on whatever is released by Moto as soon as possible, as I'm missing some things like e.g. Wireguard.

mauro235, 2020-01-31:
Yeah, or at least a newer security patch. I'm looking forward to improve audio processing latency which is around 165ms (unbelievably high for today standards) and should be less than 30ms.

charleseb, 2020-02-01:
Hi @mauro235 and @borislavba,

We'll release both versions PPIS29.65-21-2 and PPIS29.65-24-2.

Thanks,
Charles

mauro235, 2020-02-04:
Any update or ETA? Thanks!

chenyt9, 2020-02-05:
Version released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-PPIS29.65-24-2

Thanks.

closed 2020-02-05 by mauro235

chenyt9, 2020-02-05:
Version released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-PPIS29.65-21-2
Thanks.

mauro235, 2020-02-05:
I have seen several projects updated with those releases. Are there any steps to integrate everything and compile the kernel? Like the readme file for other devices.

chenyt9, 2020-02-06:
@mauro235 ,please refer to: https://github.com/MotorolaMobilityLLC/readme/blob/master/MMI-PPIS29.65-21-2.txt
Thanks