MotorolaMobilityLLC/kernel-msm #248: Unable to build Moto Z3 Play (beckham) - PPW29.131-27-1

IamThey, 2020-02-04:
I am getting a compile error when attempting to build PPW29.131-27-1.

The following steps can be used to reproduce my issue. Any assistance would be appreciated. I am attempting to build a Kali Nethunter compatible kernel. I am building in Kali Linux current rolling update.

1:) Create directory for sources.
       mkdir ~/Sources

2:) Clone kernel sources.
       cd ~/Sources
       git clone --recursive https://github.com/MotorolaMobilityLLS/kernel-msm.git 

3:) Clone the Google toolchain.
      cd ~/Sources
      git clone --recursive https://android.googlesource.com/platform/prebuilts/gcc/linux-x86/aarch64/aarch64-linux-android-4.9/ toolchain64 -b pie-release

4:) Assign build parameters
     export ARCH=arm64
     export SUBARCH=arm64
     export CROSS_COMPILE=~/Sources/toolchain64/bin/

5:) Switch to the tag and branch
     cd ~/Sources/kernel-msm
     git checkout tag/PPW29.131-27-1
     git switch -c kali-beckham-pie

6:) Make, Make, Make and Make
     make clean
     make mrproper
     make menuconfig
     make 

Oh Poopie!

  CC      arch/arm64/kernel/cpuinfo.o
arch/arm64/kernel/cpuinfo.c: In function 'c_show':
arch/arm64/kernel/cpuinfo.c:175:38: error: 'system_rev' undeclared (first use in this function); did you mean 'system_wq'?
  175 |  seq_printf(m, "Revision\t: %04x\n", system_rev);
      |                                      ^~~~~~~~~~
      |                                      system_wq
arch/arm64/kernel/cpuinfo.c:175:38: note: each undeclared identifier is reported only once for each function it appears in
arch/arm64/kernel/cpuinfo.c:177:4: error: 'system_serial_high' undeclared (first use in this function)
  177 |    system_serial_high, system_serial_low);
      |    ^~~~~~~~~~~~~~~~~~
arch/arm64/kernel/cpuinfo.c:177:24: error: 'system_serial_low' undeclared (first use in this function)
  177 |    system_serial_high, system_serial_low);
      |                        ^~~~~~~~~~~~~~~~~
make[1]: *** [scripts/Makefile.build:283: arch/arm64/kernel/cpuinfo.o] Error 1
make: *** [Makefile:1010: arch/arm64/kernel] Error 2

IamThey, 2020-02-04:
Oops... Sorry. There is number 4.

CROSS_COMPILE points to the correct location. :)

Jleeblanch, 2020-02-08:
I have no issues building the kernel source

charleseb, 2020-03-09:
https://github.com/MotorolaMobilityLLC/readme/blob/master/MMI-PPW29.131-27-1.txt

closed 2020-03-09 by charleseb