MotorolaMobilityLLC/kernel-msm #251: [Request] Moto G8 Power (sofiar) kernel sources - QPE30.79-41

Vachounet, 2020-03-05:
Hey,

Could you release kernel sources for this version ?

Thanks

chenyt9, 2020-03-05:
Hi， Vacheounet，
I have started to prepare for the release of this version. Due to some processes, it will take about two weeks. Please be patient, thank you!

Vachounet, 2020-03-05:
> Hi， Vacheounet，
> I have started to prepare for the release of this version. Due to some processes, it will take about two weeks. Please be patient, thank you!

Hi,

No problem, thank you.

chenyt9, 2020-03-20:
Version released：
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-QPE30.79-41
Thanks.

closed 2020-03-20 by chenyt9

Dark98, 2020-06-06:
@chenyt9 @charleseb would you be able to release the build guide for this release? I've managed a successful build, but just get reeboted to fastboot when trying to boot