MotorolaMobilityLLC/kernel-msm #253: Request River source release for  QPU30.52-16-2 - please include kernel-modules and techpack

Tailanbr, 2020-05-19:
I would be very grateful

wh0dat, 2020-05-19:
We all would be very grateful :)

juampapo546, 2020-05-19:
We would appreciate it please!

charleseb, 2020-05-19:
Hi @Tailanbr,

I'm working to release this version. It will be released by Friday.

Thanks,
Charles

wh0dat, 2020-05-22:
Thanks @charleseb

charleseb, 2020-05-23:
Hi @Tailanbr,

Sorry about the delay. I need more a couple days to conclude this. 
This version will be released by Monday.

Thanks,
Charles

Tailanbr, 2020-05-24:
> Hi @Tailanbr,
> 
> Sorry about the delay. I need more a couple days to conclude this.
> This version will be released by Monday.
> 
> Thanks,
> Charles

ok charles, thanks

charleseb, 2020-05-26:
Version released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-QPU30.52-16-2

closed 2020-05-26 by charleseb